RobotWebTools/ros2-web-bridge #1: ros2bridge kick-off

minggangw, 2017-10-25:
Hi @jihoonl, as [rclnodejs](https://github.com/RobotWebTools/rclnodejs) is nearly done, we want to move forward to start the **ros2bridge** project. Generally, we want to follow the rosbridge v2 Protocol and leverage the `rclnodejs`, our aim is to influence the `roslibjs` as little as possible, thus  a series of components, which are based on `roslibjs`, could be available to avoid huge effort on porting to ros2bridge. Do you have any suggestions or thing we can do to promote the `rclnodejs` client? Thanks!

jihoonl, 2017-10-25:
thank you for awesome works in `rclnodejs`. There is not so many things I can suggest at the moment. How about having an announcement in [ng-ros discourse](https://discourse.ros.org/c/ng-ros) to grab ros2 developers attention? I think they can give you some opinions about `rclnodejs` as well as `ros2bridge`.

minggangw, 2017-10-25:
We are planning to have an alpha release of `rclnodejs` recently, I think we can make an announcement in ros discourse after that milestone.

jihoonl, 2017-10-25:
Cool! Looking forward to it. :smile:

kenny-y, 2017-10-26:
👍

minggangw, 2017-11-03:
We just made the `rclnodejs` announcement at https://discourse.ros.org/t/node-js-client-for-ros-2-0-is-ready/3096, thanks!

minggangw, 2017-12-21:
This first patch has landed(6ba7c515d16d079), @jihoonl would you please help me set the `develop` branch as the default one?

jihoonl, 2017-12-21:
Hi @minggangw, now ROS2.0 Team has a full access to rclnodejs and ros2bridge_suite-experiment. Feel free to rename the repository as something-else that you prefer. e.g) `rclbridge-suite` or `rclwebbridge-suite`.

kenny-y, 2017-12-22:
+1 for `rclwebbridge-suite` naming 👍

minggangw, 2017-12-28:
Kenny and I decided to rename the repo to `ros2-web-bridge`, which stands for a bridge for ROS 2.0 and web specific.