RobotWebTools/ros2-web-bridge #4: Add Authentication feature

kenny-y, 2017-12-21:
> op `auth` - optional authentication information can be passed via the rosbridge protocol to authenticate a client connection. This information should come from some trusted third-party authenticator.

kenny-y, 2017-12-22:
There are 2 ways to do it based on preliminary analysis: 
1. wait for ROS 2.0 equivalent of `rosauth` and call into it
1. write the similar logic of `rosauth` ([< 150 lines of C++ code](https://github.com/GT-RAIL/rosauth/blob/develop/src/ros_mac_authentication.cpp)) in JavaScript

@minggangw WDYT?

minggangw, 2017-12-22:
I look through the features of the next [release](https://github.com/ros2/ros2/wiki/Roadmap), which will happen in summer 2018, `security improvements` section mentions the authentication.