RobotWebTools/ros2-web-bridge #26: Status message response should be sent back to clients for every message

qiuzhong, 2018-01-19:
This project implemented the  [rosbridge v2 protocol](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md)
Generally, sending an rosbridge message to the bridge should always return a response no matter the message is handled well.

However, this is not the cases, web socket client cannot receive response from the bridge web socket server:

```
const webSocket = require('ws');
const ws = new webSocket('ws://localhost:9090');

ws.on('open', function open() {
  ws.send('{"op": "advertise", ... }');
});

ws.on('message', function incoming() {
  // never enter
});
```

To make the rosbridge v2 protocol testing easier, please implement this feature.

qiuzhong, 2018-01-19:
This is also the experimental feature in [3.2 section of rosbridge v2 protocol](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#32-status-messages)

minggangw, 2018-01-19:
Thanks for the finding, I will take this issue.

closed 2018-01-22 by minggangw

qiuzhong, 2018-01-22:
Verified. Basically valid operations can receive responses now.

PR #29 (2018-01-22): Send back the operation status to client
Currently, we only send back the status  when there is an error. Here we should
follow the protocol
https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md

Whatever the result is, we will return the status of the operation. 'level' equals to
'error' when error happens, 'level' equals to 'none' when the execution is successful.

Fix #26
commit d0005d88 (2018-01-22): Send back the operation status to client

Currently, we only send back the status  when there is an error. Here we should
follow the protocol
https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md

Whatever the result is, we will return the status of the operation. 'level' equals to
'error' when error happens, 'level' equals to 'none' when the execution is successful.

Fix #26