RobotWebTools/ros2-web-bridge #33: Error in npm install, please suggest the possible solution.

prasun2712, 2018-01-23:
-
> ref@1.3.5 install /home/username/ros2-web-bridge/node_modules/rclnodejs/node_modules/ref
> node-gyp rebuild

make: Entering directory '/home/username/ros2-web-bridge/node_modules/rclnodejs/node_modules/ref/build'
  CXX(target) Release/obj.target/binding/src/binding.o
../src/binding.cc: In function ‘void init(Nan::ADDON_REGISTER_FUNCTION_ARGS_TYPE)’:
../src/binding.cc:643:8: warning: ‘Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’ is deprecated [-Wdeprecated-declarations]
   Nan::ForceSet(target, Nan::New<v8::String>("endianness").ToLocalChecked(), Nan::New<v8::String>(CheckEndianness()).ToLocalChecked(), static_
        ^
In file included from ../node_modules/nan/nan.h:197:0,
                 from ../src/binding.cc:7:
../node_modules/nan/nan_maybe_43_inl.h:116:35: note: declared here
 NAN_DEPRECATED inline Maybe<bool> ForceSet(
                                   ^
../src/binding.cc:643:8: warning: ‘Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’ is deprecated [-Wdeprecated-declarations]
   Nan::ForceSet(target, Nan::New<v8::String>("endianness").ToLocalChecked(), Nan::New<v8::String>(CheckEndianness()).ToLocalChecked(), static_
        ^
In file included from ../node_modules/nan/nan.h:197:0,
                 from ../src/binding.cc:7:
../node_modules/nan/nan_maybe_43_inl.h:116:35: note: declared here
 NAN_DEPRECATED inline Maybe<bool> ForceSet(
                                   ^
../src/binding.cc:643:187: warning: ‘Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’ is deprecated [-Wdeprecated-declarations]
 endianness").ToLocalChecked(), Nan::New<v8::String>(CheckEndianness()).ToLocalChecked(), static_cast<PropertyAttribute>(ReadOnly|DontDelete));
                                                                                                                                             ^
In file included from ../node_modules/nan/nan.h:197:0,
                 from ../src/binding.cc:7:
../node_modules/nan/nan_maybe_43_inl.h:116:35: note: declared here
 NAN_DEPRECATED inline Maybe<bool> ForceSet(
                                   ^
../src/binding.cc:644:8: warning: ‘Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’ is deprecated [-Wdeprecated-declarations]
   Nan::ForceSet(target, Nan::New<v8::String>("NULL").ToLocalChecked(), WrapNullPointer(), static_cast<PropertyAttribute>(ReadOnly|DontDelete))
        ^
In file included from ../node_modules/nan/nan.h:197:0,
                 from ../src/binding.cc:7:
../node_modules/nan/nan_maybe_43_inl.h:116:35: note: declared here
 NAN_DEPRECATED inline Maybe<bool> ForceSet(
                                   ^
../src/binding.cc:644:8: warning: ‘Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’ is deprecated [-Wdeprecated-declarations]
   Nan::ForceSet(target, Nan::New<v8::String>("NULL").ToLocalChecked(), WrapNullPointer(), static_cast<PropertyAttribute>(ReadOnly|DontDelete))
        ^
In file included from ../node_modules/nan/nan.h:197:0,
                 from ../src/binding.cc:7:
../node_modules/nan/nan_maybe_43_inl.h:116:35: note: declared here
 NAN_DEPRECATED inline Maybe<bool> ForceSet(
                                   ^
../src/binding.cc:644:142: warning: ‘Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’ is deprecated [-Wdeprecated-declarations]
  Nan::ForceSet(target, Nan::New<v8::String>("NULL").ToLocalChecked(), WrapNullPointer(), static_cast<PropertyAttribute>(ReadOnly|DontDelete));
                                                                                                                                             ^
In file included from ../node_modules/nan/nan.h:197:0,
                 from ../src/binding.cc:7:
../node_modules/nan/nan_maybe_43_inl.h:116:35: note: declared here
 NAN_DEPRECATED inline Maybe<bool> ForceSet(
                                   ^
  SOLINK_MODULE(target) Release/obj.target/binding.node
  COPY Release/binding.node
make: Leaving directory '/home/username/ros2-web-bridge/node_modules/rclnodejs/node_modules/ref/build'

> rclnodejs@0.3.1 install /home/username/ros2-web-bridge/node_modules/rclnodejs
> node-gyp rebuild

module.js:327
    throw err;
    ^

Error: Cannot find module 'nan'
    at Function.Module._resolveFilename (module.js:325:15)
    at Function.Module._load (module.js:276:25)
    at Module.require (module.js:353:17)
    at require (internal/module.js:12:17)
    at [eval]:1:1
    at Object.exports.runInThisContext (vm.js:54:17)
    at Object.<anonymous> ([eval]-wrapper:6:22)
    at Module._compile (module.js:409:26)
    at node.js:579:27
    at nextTickCallbackWith0Args (node.js:420:9)
gyp: Call to 'node -e "require('nan')"' returned exit status 1 while in binding.gyp. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:305:16)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 4.13.0-26-generic
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/username/ros2-web-bridge/node_modules/rclnodejs
gyp ERR! node -v v4.4.2
gyp ERR! node-gyp -v v3.3.1
gyp ERR! not ok 
npm ERR! Linux 4.13.0-26-generic
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install"
npm ERR! node v4.4.2
npm ERR! npm  v2.15.0
npm ERR! code ELIFECYCLE

npm ERR! rclnodejs@0.3.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the rclnodejs@0.3.1 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the rclnodejs package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs rclnodejs
npm ERR! Or if that isn't available, you can get their info via:
npm ERR! 
npm ERR!     npm owner ls rclnodejs
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/username/ros2-web-bridge/npm-debug.log

qiuzhong, 2018-01-24:
Please report your Node.js version. We've tried on `Node.js v8.9.4 (with npm 5.6.0)` and it worked.

minggangw, 2018-01-24:
I can reproduce this problem on node `v4.4.2`, it was caused by not installing the `nan` module, although `nan` is a dependency of other modules already. On a later version of node.js, it seems workable. I will update the `dependencies` section of [package.json](https://github.com/RobotWebTools/rclnodejs/blob/develop/package.json#L31) to fix this version specific bug.

@prasun2712 as this bug comes from [rclnodejs](https://www.npmjs.com/package/rclnodejs), you have to wait the new release of it on npmjs.com. If you want to try it immediately, I suggest two methods:

* Upgrade your node.js
* Add `"nan": "^2.8.0"` to your package.json of `rclnodejs` module.

Thanks for your finding!

qiuzhong, 2018-01-25:
Suggest to upgrade Node.js version as `rclnodejs` doesn't support for Node.js 4.x series. Your Node.js version should not be lower than `6.10.0`.

minggangw, 2018-01-29:
The [latest release](https://github.com/RobotWebTools/rclnodejs/releases/tag/0.3.2-beta.3) of rclnodejs has resolved this problem. If you install a unqualified version of Node.js, you will receive a warning which indicates to upgrade your Node.js. Close this issue.

closed 2018-01-29 by minggangw

prasun2712, 2018-01-29:
@minggangw Thank you for the help. It is working fine with node v8.9.4 and now I am able to start ros2_web_bridge module and the express server. I can also hit the publisher.html page and it shows output as connected and is showing publish message hello from rosbridge, but topic is not visible using ros2cli (ros2 topic list). What is the possible reason and solution for same.

minggangw, 2018-01-30:
What is the command line you executed, do you try the same command when using `rclpy` client to see the topics?

minggangw, 2018-01-30:
@qiuzhong would you help me to verify it by using [ros2cli](https://github.com/ros2/ros2cli)? thanks!

qiuzhong, 2018-01-30:
ros2 topic is visible for me.
```
$ ros2 topic list
/example_topic
```

I wrote a subscription based on `rclnodejs` and it could receive the message from `publisher.html` as well.

minggangw, 2018-01-30:
@qiuzhong thanks for your support, so @prasun2712 did you do the same as @qiuzhong did? If you still meet the problem, please feedback thanks!

prasun2712, 2018-01-30:
[image]
[image]
I ran two commands from ros2-web-bridge, "node bin/rosbridge.js" and "cd examples && node index.js" and then navigated to "http://localhost:3000/html/publisher.html" but still am not able to see the topic using (ros2 topic list). I have attached the images for the same.

minggangw, 2018-01-31:
I double checked my execution result of `ros2 topic list`, it works fine. 
[image]

Please check whether it works when you are running a rclnodejs [example](https://github.com/RobotWebTools/rclnodejs/blob/develop/example/publisher-example.js) or a rclpy [example](https://github.com/ros2/examples/tree/master/rclpy/topics/minimal_publisher), maybe the `ros2 topic` has something wrong.

prasun2712, 2018-01-31:
rclnodejs example "publisher-example.js" returns error : 

(node:12179) UnhandledPromiseRejectionWarning: Unhandled promise rejection (rejection id: 1): Error: The message required does not exist: std_msgs, msg, String
(node:12179) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code

Is this error with my rclnodejs cos rcl is working fine when I am publishing using terminal and the trying to echo the same topic?

minggangw, 2018-01-31:
It seems that the generated `.js` files are missing, you can delete the `generated` folder manually and run the example again, all the message files will be regenerated again for you.

prasun2712, 2018-01-31:
Thanks @minggangw , so rclnodejs examples are working for me, but ros2-web-bridge examples index.js is not working even though the html page shows its connected and is publishing on the topic "example_topic".

minggangw, 2018-01-31:
Currently the `roslibjs` running in your browser doesn't check the result of the execution, I suggest you can write a Node.js app by `rclnodejs` to see if you can receive the topic published from the browser.

Also you can check the `rclnodejs` module under your `node_modules`, and run the example.

prasun2712, 2018-01-31:
Sure, will do that and get back to you in case of any problem. :) Thanks buddy.

prasun2712, 2018-02-01:
Hi @minggangw ! I did all the testing. My findings are : 
1) Using rclnodejs examples I am able to publish and subscribe to topics and I am able to visualize the topics and the contents using rcl in terminal.
2) ros2-web-bridge is up but the html page is not publishing on the topic using roslibjs neither it is subscribing to the published topic.
I am not sure what exactly is the issue, also is there any way to run ros2-web-bridge in verbose mode?

minggangw, 2018-02-02:
You can run

```bash
DEBUG=ros2-web-bridge* node bin/rosbridge.js
```

to see the debug information.

prasun2712, 2018-02-02:
I am getting error , The message required does not exist. Also I am not getting message that web socket server started on "IP:PORT".

[image]

minggangw, 2018-02-02:
It seems that you meet the same problem as you were running rclnodejs examples. I estimate the reason why you have it is that you switched the version of Node.js without installing the dependencies again. So I suggest:

```bash
rm -rf node_modules/
npm install
```
This will clean up the environment, hope it works!

prasun2712, 2018-02-02:
Yes deleting node_modules folder and clearing cache (npm cache clear --force), then installing again did the work for me. Thank you. But the warning related to "NAN_DEPRECATED" still came though I am using node version (v8.9.4).

minggangw, 2018-02-02:
That warning comes from [NAN](https://github.com/nodejs/nan) when compiling, which will not lead to problems, so just go ahead!

prasun2712, 2018-02-23:
Hi @minggangw ! When I am subscribing a topic via .html page and then after receiving the message if I am unsubscribing it, I am getting error as "ros2-web-bridge:RefCountingHandle Handle is destroyed. +0ms" and then again reloading the page I am not receiving the message, also I am not getting the message "ros2-web-bridge:Bridge Send message to subscription." from ros2_web_bridge end which I got when I first subscribed to the message. I have attached the screenshot for the same. Please suggest.
[image]

minggangw, 2018-02-23:
> I am getting error as "ros2-web-bridge:RefCountingHandle Handle is destroyed. +0ms" 

Actually this is not an error. When you unsubscribe a topic, the **bridge** will destroy the practical handle created by `rclnodejs` if this handle is not occupied by others any longer. If you reload the page,  a subscriber will be created again which is transparent to the caller.

I saw a weird output from the log,  **JSON command received: 14**, which means the bridge has received a command which can not be recognized and the connection gets closed. So I think that is the main reason why you can't receive the subscribed messages. Can you reproduce this problem stably?

prasun2712, 2018-02-26:
That error **JSON command received: 14** is not there this time but the issue is same, when I am unsubscribing the topic I can see that handle is destroyed (That is correct as you explained.), but after reloading the page same JSON message with "op" code subscribe and same other fields is being sent to ros2 bridge but it is not sending the message to subscription as you can see that line is not printed in the output.
1st Image
=================
[image]
2nd Image
=================
[image]
**Note** : I have added a line "example.unsubscribe()" in "subscription.html" example provided by you to unsubscribe the message after receiving single message.

minggangw, 2018-02-26:
[subscription.html.zip](https://github.com/RobotWebTools/ros2-web-bridge/files/1757590/subscription.html.zip)
Please test with this HTML page to replace the one under the `examples` folder`.

prasun2712, 2018-02-26:
Same issue with the page you sent. On clicking the unsubscribe button, the message stops, handle is destroyed and upon reloading the page I am not getting the messages even though the connection is there with the bridge and it is receiving the JSON message for subscription. I have to restart the bridge again for it to work.

minggangw, 2018-02-26:
It's a little weird, as it works fine on my desktop with the same page. Would you please use

```
ros2 topic echo
```

to check the topic you published is there?

prasun2712, 2018-02-26:
Yes, That I have checked. Topic is being continuously published.

minggangw, 2018-02-26:
@prasun2712 Are you using the latest version of [ros2-web-bridge v0.2.1](https://www.npmjs.com/package/ros2-web-bridge), and through the log I didn't see the `publisher`'s output, which client do you use for the publisher, rclpy or rclnodejs?

@qiuzhong would you please verify this problem to see if it can be reproduced? thanks!

prasun2712, 2018-02-26:
Version is "0.2.0" and I am publishing using ros2 command line tool i.e. " ros2 topic pub -r 5 /example_topic std_msgs/String "data: Hello world" "

minggangw, 2018-02-26:
I was testing with v0.2.1 and the `html/publisher.html` and `html/subscription.html`, I think it doesn't matter that you were using the `ros2cli` tools. So please try to update your ros2-web-bridge.

prasun2712, 2018-02-26:
Sure, will do that and test.

qiuzhong, 2018-02-27:
@minggangw , I didn't reproduce this issue for both `rclnodejs` client and `ros2 topic pub -r 5 /example_topic std_msgs/String "data: Hello world"` command line.

minggangw, 2018-02-27:
Thanks @qiuzhong, let's wait the feedback from @prasun2712 .