RobotWebTools/ros2-web-bridge #36: rosbridge v2 protocol specification: advertise

qiuzhong, 2018-01-24:
According to the [rosbridge v2 protocol specification](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#341-advertise--advertise-)

> If the topic already exists with a different type, an error status message is sent and this message is dropped.

```
// 1st message
Message: '{"op": "advertise", "id": "create_topic1", "topic": "example_topic", "type": "std_msgs/String"}'
Expected: success
Actual response: '{"op": "set_level", "level": "none"}'

// 2nd message
Message: '{"op": "advertise", "id": "create_topic2", "topic": "example_topic", "type": "std_msgs/Char"}'
Expected: error as same topic already exists but with different type
Actual response: '{"op": "set_level", "level": "none"}'
```

minggangw, 2018-01-24:
I will check this situation to see if it's supported by ROS2 now.

closed 2018-02-02 by minggangw

qiuzhong, 2018-02-05:
Tried with PR42. Same as PR43, one more normal response is sent back to the client.

qiuzhong, 2018-02-06:
Verified. No more the extra response any more.

PR #42 (2018-02-02): Return error when the topic has already existed
To align with the rosbridge v2 protocol spec:

- If the topic already exists with a different type, an error status message
  is sent and this message is dropped.

This patch implements this logic. Firstly it will check if the topic exists,
then check the type of it if there is already one. Finally, if the type is
different from the previous one and an error is returned.

Fix #36
commit 7fbfc79a (2018-02-02): Return error when the topic has already existed

To align with the rosbridge v2 protocol spec:

- If the topic already exists with a different type, an error status message
  is sent and this message is dropped.

This patch implements this logic. Firstly it will check if the topic exists,
then check the type of it if there is already one. Finally, if the type is
different from the previous one and an error is returned.

Fix #36