RobotWebTools/ros2-web-bridge #41: Failed to launch rosbridge.js server on Windows

qiuzhong, 2018-02-01:
I can install `ros2-web-bridge` on Windows successfully. However, it failed to launch `bin\rosbridge.js`:
```
call /path/to/ros2-windows/local_setup.bat
git clone https://github.com/RobotWebTools/ros2-web-bridge.git
cd ros2-web-bridge
npm install   // succeed by far

node bin\ros2-bridge.js
C:\dev\ros2\ros2-web-bridge\node_modules\bindings\bindings.js:88
        throw e
        ^

Error: The specified module could not be found.
\\?\C:\dev\ros2\ros2-web-bridge\node_modules\rclnodejs\build\Release\rclnodejs.node
    at Object.Module._extensions..node (module.js:672:18)
    at Module.load (module.js:556:32)
    at tryModuleLoad (module.js:499:12)
    at Function.Module._load (module.js:491:3)
    at Module.require (module.js:587:17)
    at require (internal/module.js:11:18)
    at bindings (C:\dev\ros2\ros2-web-bridge\node_modules\bindings\bindings.js:81:44)
    at Object.<anonymous> (C:\dev\ros2\ros2-web-bridge\node_modules\rclnodejs\lib\node.js:17:38)
    at Module._compile (module.js:643:30)
    at Object.Module._extensions..js (module.js:654:10)
```

## Environments:
* Windows 10 1607
* Node.js: 8.9.4
* VS2017 Community Edition

qiuzhong, 2018-02-02:
I checked the Node.js addon file
`C:\dev\ros2\ros2-web-bridge\node_modules\rclnodejs\build\Release\rclnodejs.node` and it actually exists.

minggangw, 2018-02-02:
We need a version for windows to start our module.

minggangw, 2018-02-09:
Hi @qiuzhong I tested this issue on my windows laptop and it seems workable. I met the same error at the beginning, but soon I noticed that you have to call the

```
> call C:\dev\ros2\local_setup.bat
```
every time if you run `refreshenv`. So would you please try it again? if there is no problem, I will close this issue and #39. We will use `rosbridge.js` for Linux&Windows.

minggangw, 2018-02-12:
Hi @qiuzhong would you please verify this issue, and if it works, we can release a bug fix edition recently. (We have landed several commits to fix the rosbridge protocol issues, I think it is the time to make a minor release)

qiuzhong, 2018-02-12:
Sorry for late response.

By using the recent packages, I cannot reproduce this issue any more. This issue is reported when I got the ros2 release package that was before the `rcl` API change: [ros2/rcl@44801b3](https://github.com/ros2/rcl/commit/44801b398e6f0). At present, those old ros2 packages are not compatible the latest `rclnodejs` dependency. 

So I don't think this issue is still valid. @minggangw , I suggest to close this issue and let's start the release.

minggangw, 2018-02-12:
Thanks for your verification! I will remove the `rosbridge.bat` file then and close issue #39 and this one, thanks!

minggangw, 2018-02-12:
Close this issue per comments above.

closed 2018-02-12 by minggangw