RobotWebTools/ros2-web-bridge #45: ros2-bridge protocol: advertise operation with same topic and type

qiuzhong, 2018-02-06:
According to the [rosbridge v2.0 Protocol Specification section 3.4.1](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#341-advertise--advertise-)

> If the topic already exists with the same type, the sender of this message is registered as another publisher.

Does this mean if a second `advertise` with the same topic and type, the current publisher should be destroyed and create a new one?
If so, what's the point?

qiuzhong, 2018-02-06:
@minggangw , please take a look.

If I didn't get wrong, the ros2-web-bridge implementation only keep one publisher for multiple advertise operations with same topic and type. I think this is a good solution as only one publisher instance was created. I'm confused by the spec.

minggangw, 2018-02-06:
Your understanding of the implementation of `ros2-web-bridge` is right. If a publisher with a existing topic and the same type is going to be created, the `ros2-web-bridge` will not create another one practically and the requested publisher will get backed by the existing one.

The description of this situation is not clear in the [spec](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#341-advertise--advertise-), because we can not distinguish two publishers which have the same `topic` and `message type` through the current JSON commands. I think the spec wants to express a policy that although there is already a publisher with the same topic and message type, it should be created successfully for the second one. The way of implementing depends on the developer himself.

qiuzhong, 2018-02-07:
I think the current implementation is just fine though it doesn't comply the spec strictly.

minggangw, 2018-02-09:
So, I think we can close this issue now, thanks!

closed 2018-02-09 by minggangw