RobotWebTools/ros2-web-bridge #46: Invalid topic type for unadvertise operation should return error not warning

qiuzhong, 2018-02-07:
For `advertise` operation, invalid topic will cause an error status response as an TypeError is caught during the `executeCommand`. However, invalid topic type for `unadvertise` operation will return a warning status response, like this:

```
    this._registerOpMap('unadvertise', (command) => {
      if (!this._topicsPublished.has(command.topic)) {
        debug(`The topic ${command.topic} does not exist.`);
        let error = new Error();
        error.level = 'warning';
        throw error;
      }
      debug(`unadvertise a topic: ${command.topic}`);
      this._topicsPublished.delete(command.topic);
      this._resourceProvider.destroyPublisher(command.topic);
    });
```
Since the topic is not in `_topicPublished`, it simply return a warning status response.

Well, the specification doesn't indicate this explicitly but I think it should keep aligned with the `advertise` operation.

minggangw, 2018-02-07:
We should distinguish these two case:
- invalid topic
- doesn't exist topic

With different level of errors, I will investigate this issue.

closed 2018-02-09 by minggangw

qiuzhong, 2018-02-09:
Verified. This issue is fixed.

PR #48 (2018-02-09): Validate the topic when executing unadvertise
When the rosbridge is going to unadvertise a topic, the topic should be
validated first. If it is invalid, an error should be returned.

Fix #46
commit 7ee56fac (2018-02-09): Validate the topic when executing unadvertise

When the rosbridge is going to unadvertise a topic, the topic should be
validated first. If it is invalid, an error should be returned.

Fix #46