RobotWebTools/ros2-web-bridge #49: Rosbridge v2 protocol: publish with subset msg of type of topic

qiuzhong, 2018-02-08:
In [rosbridge v2 protocol section 3.4.3 - publish](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#343-publish--publish-)

> If the msg is a subset of the type of the topic, then a warning status message is sent and the unspecified fields are filled in with defaults

Let's say we advertise a topic with type `std_msgs/Header` (contains a `builtin_interface/Time` field and a `std_msgs/String` field) and send a `Time` type data. The response status is `none` which indicates the operation is successful.

```
Header.msg:
builtin_interfaces/Time stamp
string frame_id

Time.msg:
int32 sec
uint32 nanosec
 
Request: {"op": "adverstie", "topic": "example_topic", "type": "std_msgs/Header"}
Response: {"op": "set_level", "level": "none"}

Request: {"op": "publish", "topic": "example_topic", "msg": {stamp: {sec: 123456, nanosec: 789}}}
Response: {"op": "set_level", "level": "none"}
```

Actually, this is because `rclnodejs` doesn't complain if the message to be publish is a subset of the type of the topic. For `rclnodejs` the unspecified fields are filled with garbage data.

minggangw, 2018-02-09:
This task should be handled by `rclnodejs` internally, https://github.com/RobotWebTools/rclnodejs/issues/243 is related with this issue.

qiuzhong, 2018-02-09:
So `rclnodejs` will return a warning or something in that case? Even `rclnodejs` handle with the unspecified fields of the message type, bridge should also catch this exception and return a `warning` status response.

minggangw, 2018-02-09:
Unfortunately, `rclnodejs` will not complain.

minggangw, 2018-02-27:
Hi @qiuzhong as https://github.com/RobotWebTools/rclnodejs/commit/925966b6af207b401c4d2a8b216b46f604929fdc has landed, would you please re-run the CIs of #57, thanks!

minggangw, 2018-02-27:
Sorry, I forgot we have to update the package of `rclnodejs` on the npmjs, so let's release a new edition of it first.

qiuzhong, 2018-02-27:
@minggangw , I just re-run the test cases and get an error response while the spec asks for an warning.

minggangw, 2018-02-28:
Do you have the log?

qiuzhong, 2018-02-28:
@minggangw ,  here is the bridge server log:
```
$ node bin/rosbridge.js 
  ros2-web-bridge:index The ros2-web-bridge has started. +0ms
The web socket server started on ws://localhost:9090
  ros2-web-bridge:Bridge Web bridge a9e73037-9c5b-4a14-ad3f-e597a7dc1006 is created +0ms
  ros2-web-bridge:Bridge JSON command received: {"op":"advertise","id":"publish_advertise_setup5","topic":"publish_topic5","type":"std_msgs/String"} +3ms
  ros2-web-bridge:Bridge advertise a topic: publish_topic5 +0ms
  ros2-web-bridge:ResourceProvider Publisher has been created, and the topic name is publish_topic5. +0ms
  ros2-web-bridge:Bridge Response: {"op":"set_level","level":"none"} +7ms
  ros2-web-bridge:Bridge JSON command received: {"op":"publish","id":"publish_id5","topic":"publish_topic5","msg":{"stamp":{"sec":123456,"nanosec":789}}} +1ms
  ros2-web-bridge:Bridge Publish a topic named publish_topic5 with {"stamp":{"sec":123456,"nanosec":789}} +1ms
  ros2-web-bridge:Bridge Exception caught in Bridge.executeCommand(): TypeError: Invalid argument: data in String +0ms
  ros2-web-bridge:Bridge Error: TypeError: Invalid argument: data in String happened when executing command publish +0ms
  ros2-web-bridge:Bridge Response: {"op":"set_level","id":"publish_id5","level":"error"} +0ms
  ros2-web-bridge:RefCountingHandle Handle is destroyed. +0ms
  ros2-web-bridge:Bridge Web bridge a9e73037-9c5b-4a14-ad3f-e597a7dc1006 is closed +3ms
```

And here is the log from web socket client:
```
sending request: {"op":"advertise","id":"publish_advertise_setup5","topic":"publish_topic5","type":"std_msgs/String"}
1:{"op":"set_level","level":"none"}
sending request: {"op":"publish","id":"publish_id5","topic":"publish_topic5","msg":{"stamp":{"sec":123456,"nanosec":789}}}
2:{"op":"set_level","id":"publish_id5","level":"error"}
```

qiuzhong, 2018-02-28:
Sorry, I just used the wrong message type in `advertise` operation above. I corrected it and updated the log. As you see, it still returned an error response.

minggangw, 2018-02-28:
I see, we have to catch this exception in `ros2-web-bridge` in order to return the warning message to the browser. I will submit a PR to fix it.

minggangw, 2018-03-01:
rosbridge v2.0 Protocol Specification about this case is:
> If the msg is a subset of the type of the topic, then a warning status message is sent and the unspecified fields are filled in with defaults

According to the current implementation of rclnodejs https://github.com/RobotWebTools/rclnodejs/commit/925966b6af207b401c4d2a8b216b46f604929fdc, there will be no **default** value if the subset is not specified. So if we return a warning here, actually the topic is not published which I think it not reasonable because we have dropped it.

Due to the specific traits of ROS 2, I prefer to maintain a table of incompatibility of the current implementation (cases like, #59 and the one recorded in [README](https://github.com/RobotWebTools/ros2-web-bridge#known-issues) , what's your opinion @qiuzhong ?

qiuzhong, 2018-03-01:
Agreed. Due to the implementation limits, we should let the future users know this.

PR #60 (2018-03-01): Add the compability section in README
Fix #49,#59
commit 4d9a0efa (2018-03-01): Add the compability section in README

Fix #49,#59
closed 2018-03-01 by minggangw