RobotWebTools/ros2-web-bridge #59: Message data not inferred for subscribe operation

qiuzhong, 2018-02-24:
According to [rosbridge v2 protocol section 3.4.4 subscription](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#344-subscribe)
> If left off, type will be inferred.

Let's take a look at the `subscribe` operation:
```
{ "op": "subscribe",
  (optional) "id": <string>,
  "topic": <string>,
  (optional) "type": <string>,
  (optional) "throttle_rate": <int>,
  (optional) "queue_length": <int>,
  (optional) "fragment_size": <int>,
  (optional) "compression": <string>
}
```

`type` field is optional, however, sending a `subscribe` operation without `type` field to the bridge server will get an error response.
```
Request: {op: 'subscribe', id: 'subscribe_id1', topic: 'example_topic'}
Response: {"op":"set_level","id":"subscribe_id1","level":"error"}
```

This error happened in [_exractMessageType(type)](https://github.com/RobotWebTools/ros2-web-bridge/blob/develop/lib/bridge.js#L113) before creating the `rclnodejs` subscription.

minggangw, 2018-02-26:
I am afraid that we can not keep compatible with the rosbridge v2 protocol in ROS 2. The reason is below:

In ROS 1 era, we can get the message type by calling `rostopic`, see this [example](https://github.com/RobotWebTools/rosbridge_suite/blob/ee339d54474b87aa2f56b4755cbf095a99d25bff/rosbridge_library/src/rosbridge_library/internal/subscribers.py#L36). But the [ros2cli](https://github.com/ros2/ros2cli) in ROS 2 doesn't have this interface, and we have to carry the exact message type in the request from browser. The result is what you have gotten if the request didn't have the `type` part.

minggangw, 2018-03-01:
Close this issue as we can not keep compatible with the rosbridge protocol. And we have added the compatibility section into README to point out the differences (#60).

closed 2018-03-01 by minggangw