RobotWebTools/ros2-web-bridge #63: Subscribe op implementation incompatible with rosbridge protocol spec

qiuzhong, 2018-03-05:
Here is the `publish` operation interface defined in the [spec](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#344-subscribe)
```
{ "op": "subscribe",
  (optional) "id": <string>,
  "topic": <string>,
  (optional) "type": <string>,
  (optional) "throttle_rate": <int>,
  (optional) "queue_length": <int>,
  (optional) "fragment_size": <int>,
  (optional) "compression": <string>
}
```

Though the `throttle_rate`, `queue_length` are optional fields, they will impact the subscriber's behavior. The spec requires:

> If queue_length is specified, then messages are placed into the queue before being sent. 
> If a client has multiple subscriptions to the same topic, then messages are sent at the lowest throttle_rate, with the lowest fragmentation size, and highest queue_length. 

In current implementation of rosbridge, this field was never considered for `subscribe` operation.

qiuzhong, 2018-03-05:
@minggangw , please take a look and let's discuss whether they'll be implemented or just throw them to the incompatibility table in the README.

minggangw, 2018-03-05:
There is no such a parameter like `throttle_rate` in ROS 2 if I remember it right, I will check this soon.

qiuzhong, 2018-03-05:
Right. I didn't find the `queue` keyword in rclnodejs qos policy either.

minggangw, 2018-03-22:
The parameters which do not exist in ROS 2 will not affect the functionalities of the `ros2-web-brige` and those ones will be ignored. So I think we can close this issue now.

qiuzhong, 2018-03-22:
So this will not be fixed?

minggangw, 2018-03-22:
Yes, because the current protocol is designed for ROS1 and we can not redefine it. Maybe we can have an update version in the future.

closed 2018-03-22 by qiuzhong