RobotWebTools/ros2-web-bridge #64: Complex message type altered after bridge server sending back to clients

qiuzhong, 2018-03-09:
For complex message type, like `sensor_msgs/msg/JointState`, the message object altered when the rosbridge server tried sending back it to the clients. Here it is:

* **publish sensor_msg/msg/JointState message type:**
```
{ op: 'publish', id: 'publish_setup_jointstate', topic: 'subscribe_jointstate_topic',
  msg: {
    header: {
      stamp: {
        sec: 123456, nanosec: 789
      },
      frame_id: 'main frame'},
    name: ['Tom', 'Jerry'], position: [1, 2], velocity: [2, 3], effort: [4, 5, 6]}}
```

* **Returned response**:
```
{ op: 'publish', topic: 'subscribe_jointstate_topic',
  msg: {
    header: {
        stamp: [Object],
        frame_id: 'main frame' },
     name: [ 'Tom', 'Jerry' ],
     position: { '0': 1, '1': 2 },
     velocity: { '0': 2, '1': 3 },
     effort: { '0': 4, '1': 5, '2': 6 } } }
```

The structure of the message altered after serialization.  This happened when calling `JSON.stringify()` and `JSON.parse()` of the `JointState` message object.

minggangw, 2018-03-09:
Do you mean if we create a object below

```JavaScript
let obj = { op: 'publish', id: 'publish_setup_jointstate', topic: 'subscribe_jointstate_topic',
  msg: {
    header: {
      stamp: {
        sec: 123456, nanosec: 789
      },
      frame_id: 'main frame'},
    name: ['Tom', 'Jerry'], position: [1, 2], velocity: [2, 3], effort: [4, 5, 6]}}
```

then invoke:

``` JavaScript
let str = JSON.stringify(obj);
let obj = JSON.parse(str); 
```
Finally, the structure changed?

qiuzhong, 2018-03-09:
Yes. This should be not a problem if users only use `rclnodejs` since there is no need JSON convertion. But for the bridge server, this could be a issue.

minggangw, 2018-03-09:
Yes, indeed.

minggangw, 2018-03-09:
I made a quick test and found that if you print the result of `str = JSON.stringify(obj)`, the string returned is right:
>{"op":"publish","id":"publish_setup_jointstate","topic":"subscribe_jointstate_topic","msg":{"header":{"stamp":{"sec":123456,"nanosec":789},"frame_id":"main frame"},"name":["Tom","Jerry"],"position":[1,2],"velocity":[2,3],"effort":[4,5,6]}}

you can get the value of key `nanosec` and `sec` after executing `JSON.parse(str)`.  Would you help me to check this @qiuzhong? thanks!

qiuzhong, 2018-03-09:
@minggangw , you're right. Simply make JSON conversion will not raise this.

But this issue is found when I got the data from web socket transferring of the bridge server. What I means is that when web socket clients receive the serialized message, it can be re-construct the original object any more.

qiuzhong, 2018-03-09:
Here is the bridge log:

```
 node bin/rosbridge.js 
  ros2-web-bridge:index The ros2-web-bridge has started. +0ms
The web socket server started on ws://localhost:9090
  ros2-web-bridge:Bridge Web bridge 8581766c-47e6-4966-add6-c47f22dd1595 is created +0ms
  ros2-web-bridge:Bridge JSON command received: {"op":"subscribe","id":"subscribe_jointstate","topic":"subscribe_jointstate_topic","type":"sensor_msgs/JointState"} +2ms
  ros2-web-bridge:Bridge subscribe a topic named subscribe_jointstate_topic +1ms
  rclnodejs:node Finish creating subscription, topic = subscribe_jointstate_topic. +0ms
  ros2-web-bridge:SubscriptionManager Subscription has been created, and the topic name is subscribe_jointstate_topic. +0ms
  ros2-web-bridge:Bridge Response: {"op":"set_level","level":"none"} +17ms
  ros2-web-bridge:Bridge JSON command received: {"op":"advertise","id":"advertise_setup_jointstate","topic":"subscribe_jointstate_topic","type":"sensor_msgs/JointState"} +1ms
  ros2-web-bridge:Bridge advertise a topic: subscribe_jointstate_topic +1ms
  rclnodejs:node Finish creating publisher, topic = subscribe_jointstate_topic. +3ms
  ros2-web-bridge:ResourceProvider Publisher has been created, and the topic name is subscribe_jointstate_topic. +0ms
  ros2-web-bridge:Bridge Response: {"op":"set_level","level":"none"} +0ms
  ros2-web-bridge:Bridge JSON command received: {"op":"publish","id":"publish_setup_jointstate","topic":"subscribe_jointstate_topic","msg":{"header":{"stamp":{"sec":123456,"nanosec":789},"frame_id":"main frame"},"name":["Tom","Jerry"],"position":[1,2],"velocity":[2,3],"effort":[4,5,6]}} +1ms
  ros2-web-bridge:Bridge Publish a topic named subscribe_jointstate_topic with {"header":{"stamp":{"sec":123456,"nanosec":789},"frame_id":"main frame"},"name":["Tom","Jerry"],"position":[1,2],"velocity":[2,3],"effort":[4,5,6]} +0ms
  rclnodejs:publisher Message of topic subscribe_jointstate_topic has been published. +2ms
  ros2-web-bridge:Bridge Response: {"op":"set_level","level":"none"} +1ms
  rclnodejs:subscription Message of topic subscribe_jointstate_topic received. +1ms
  ros2-web-bridge:Bridge Send message to subscription. +2ms
  ros2-web-bridge:RefCountingHandle Handle is destroyed. +0ms
  rclnodejs:node subscription destroyed. +5ms
  ros2-web-bridge:RefCountingHandle Handle is destroyed. +0ms
  ros2-web-bridge:Bridge Web bridge 8581766c-47e6-4966-add6-c47f22dd1595 is closed +4ms
...
```

minggangw, 2018-03-09:
I tested with `rclnodejs` alone and got below with `console.log(JSON.stringify(state));` in example [subscription-message-example.js
](https://github.com/RobotWebTools/rclnodejs/blob/develop/example/subscription-message-example.js#L24)

```
{"header":{"stamp":{"sec":123456,"nanosec":789},"frame_id":"main frame"},"name":["Tom","Jerry"],"position":{"0":1,"1":2},"velocity":{"0":2,"1":3},"effort":{"0":4,"1":5,"2":6}}
```

So I think the `rclnodejs` works fine, I will check the `bridge` later, thanks!

minggangw, 2018-03-09:
The literal string received by websockets is 
```
{"op":"publish","topic":"JointState","msg":{"header":{"stamp":{"sec":123456,"nanosec":789},"frame_id":"main frame"},"name":["Tom","Jerry"],"position":{"0":1,"1":2},"velocity":{"0":2,"1":3},"effort":{"0":4,"1":5,"2":6}}}
```
are you printing the result of `JSON.parse` to get the output below? 

```JavaScript
{ op: 'publish',
  topic: 'JointState',
  msg:
   { header: { stamp: [Object], frame_id: 'main frame' },
     name: [ 'Tom', 'Jerry' ],
     position: { '0': 1, '1': 2 },
     velocity: { '0': 2, '1': 3 },
     effort: { '0': 4, '1': 5, '2': 6 } } }
```

qiuzhong, 2018-03-09:
Right. That's exact what `JSON.parse()` got:

```
node
> var s = '{"op":"publish","topic":"JointState","msg":{"header":{"stamp":{"sec":123456,"nanosec":789},"frame_id":"main frame"},"name":["Tom","Jerry"],"position":{"0":1,"1":2},"velocity":{"0":2,"1":3},"effort":{"0":4,"1":5,"2":6}}}';
undefined
> s
'{"op":"publish","topic":"JointState","msg":{"header":{"stamp":{"sec":123456,"nanosec":789},"frame_id":"main frame"},"name":["Tom","Jerry"],"position":{"0":1,"1":2},"velocity":{"0":2,"1":3},"effort":{"0":4,"1":5,"2":6}}}'
> JSON.parse(s);
{ op: 'publish',
  topic: 'JointState',
  msg:
   { header: { stamp: [Object], frame_id: 'main frame' },
     name: [ 'Tom', 'Jerry' ],
     position: { '0': 1, '1': 2 },
     velocity: { '0': 2, '1': 3 },
     effort: { '0': 4, '1': 5, '2': 6 } } }
```

minggangw, 2018-03-12:
So it's clear now. The thing is that when using `console.log` to print an object of complex message, the `stamp` key of **Header** displays as `[Object]`. I thinks the output is a kind of behaviour of the `console.log` function which will not change the object itself, and it's all right here.

You can do a test to print the object defined below

```JavaScript
let obj = { op: 'publish', id: 'publish_setup_jointstate', topic: 'subscribe_jointstate_topic',
  msg: {
    header: {
      stamp: {
        sec: 123456, nanosec: 789
      },
      frame_id: 'main frame'},
    name: ['Tom', 'Jerry'], position: [1, 2], velocity: [2, 3], effort: [4, 5, 6]}}

console.log(obj);
```

and you get

```
{ op: 'publish',
  id: 'publish_setup_jointstate',
  topic: 'subscribe_jointstate_topic',
  msg: 
   { header: { stamp: [Object], frame_id: 'main frame' },
     name: [ 'Tom', 'Jerry' ],
     position: [ 1, 2 ],
     velocity: [ 2, 3 ],
     effort: [ 4, 5, 6 ] } }
```

qiuzhong, 2018-03-12:
Please note:

After web socket clients received the complex object, stamp changed from `object` to `array`, position from `array` to `object`.

qiuzhong, 2018-03-12:
We should figure out why before a publisher node sending this `JointState` type message, the `JSON.stringify` works:
```
{"header":{
    "stamp":{
      "sec":123456,
      "nanosec":789},
 "frame_id":"main frame"},
 "name":["Tom","Jerry"],
 "position":[1,2],
  "velocity":[2,3],
  "effort":[4,5,6]}
```

While a subscription node received this message, then the `JSON.stringify` didn't work:

```
{ "header":{
    "stamp":{
    "sec":123456,
    "nanosec":789},
  "frame_id":"main frame"},
  "name":["Tom","Jerry"],
  "position":{"0":1,"1":2},
  "velocity":{"0":2,"1":3},
  "effort":{"0":4,"1":5,"2":6}}
```

minggangw, 2018-03-12:
Sorry, I did not notice the `position` as an array, I did a quick check and found a possible cause here
https://github.com/RobotWebTools/rclnodejs/blob/develop/lib/message_translator.js#L143.

Thanks for finding this undetectable problem.

minggangw, 2018-03-13:
I found the reason why we got, something like `"position":{"0":1,"1":2}` here. That is because if the type of an array is primitive which may be any of 'int8', 'uint8', 'int16', 'uint16', 'int32', 'uint32', 'float64', 'float32', 'char' and 'byte', `rclnodejs` will use [**TypedArray**](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/TypedArray) to represent them. Please try the example below:

```JavaScript
const typedArray1 = new Int8Array(8);
typedArray1[0] = 32;
let str = JSON.stringify(typedArray1);
console.log(str);
console.log(JSON.parse(str));
```

and you will get the out put:

```
> "{"0":32,"1":0,"2":0,"3":0,"4":0,"5":0,"6":0,"7":0}"
> Object { 0: 32, 1: 0, 2: 0, 3: 0, 4: 0, 5: 0, 6: 0, 7: 0 }
```
 We use **TypedArray** to improve the performance, so it's a necessary implementation here.

qiuzhong, 2018-03-14:
This sounds reasonable. As when an array is converted to an object when the indexes become the keys automatically. The object can still behaves like the array. This is not perfect but it can work.

minggangw, 2018-03-14:
So let's close this issue.

closed 2018-03-14 by minggangw