RobotWebTools/ros2-web-bridge #65: Multiple subscriptions support

qiuzhong, 2018-03-09:
In terms of the subscribe-related operations, like `subscribe` and `unsubscribe`, the `id` field is to identify the subscriber. According to the rosbridge v2 protocol [subscribe](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#344-subscribe) section:

> id – if specified, then this specific subscription can be unsubscribed by referencing the ID.

and [unsubscribe](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#345-unsubscribe) section:

> If an id is provided, then only the corresponding subscription is unsubscribed. If no ID is provided, then all subscriptions are unsubscribed.

All the operations should support multiple subscriptions action. The `id` field can be useful to identify them. 

This feature should be supported by the bridge.

minggangw, 2018-03-22:
Would you please check if the `roslibjs` can support the `id` section, as `roslibjs` is the only JavaScript library we can use.

qiuzhong, 2018-03-22:
I checked the examples by launching two subscription.html instances. They have the same id when sending request to create subscription. So I guest `roslibjs` doesn't require `id` section when calling the API but it will construct the `id` field automatically.

```
{"op":"subscribe","id":"subscribe:/example_topic:1","type":"std_msgs/String","topic":"/example_topic","compression":"none","throttle_rate":0,"queue_length":0}
```

And both instances work well.