RobotWebTools/ros2-web-bridge #68: call_service operation argments incompatible with spec

qiuzhong, 2018-03-14:
According to the [rosbridge protocol call_service section](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#346-call-service)

```
{ "op": "call_service",
  (optional) "id": <string>,
  "service": <string>,
  (optional) "args": <list<json>>,
  (optional) "fragment_size": <int>,
  (optional) "compression": <string>
}
```

> args – if the service has no args, then args does not have to be provided, though an empty list is equally acceptable. Args should be a list of json objects representing the arguments to the service

There are two issues of current implementation:

```
    this._registerOpMap('call_service', (command) => {
      let serviceName = command.service;
      let client =
        this._resourceProvider.createClient(this._exractServiceType(command.args.type), serviceName);

      if (client) {
        client.sendRequest(command.args.request, (response) => {
          let serviceResponse =
            {op: 'service_response', service: command.service, values: response, id: command.id, result: true};

          this._ws.send(JSON.stringify(serviceResponse));
        });
      }
    });
```

* The spec asks for list of JSON object, however the current implementation assumes `args` as an object and access `request` property directly.

* The args never guarantees the it has the `type` property while the implementation assume it has. The bridge client may send a `call_service` request provided extra service type message field, in that case, the calling still failed as `args.type` is `undefined`.

minggangw, 2018-03-22:
This problem has been mentioned in [README](https://github.com/RobotWebTools/ros2-web-bridge#compability-with-rosbridge-v20-protocol).

qiuzhong, 2018-03-22:
Since this incompatibility is listed in the README, close this issue.

closed 2018-03-22 by qiuzhong

minggangw, 2018-03-26:
The [PR](https://github.com/RobotWebTools/roslibjs/commit/47336faa701ff1d7c3cda3f8a824ca6db62feb48) which will resolve this incompatibility has been merged, hopefully we can eliminate this issue when next release of roslibjs is on board.

qiuzhong, 2018-03-26:
So the `roslibjs` will provide the `type` field when sending a `call_service` now? So we can assume the `call_service` requests contain the `type`?

minggangw, 2018-03-26:
Yes, the `call_service` command (JSON string) will contain the information of the service type it requests.