RobotWebTools/ros2-web-bridge #69: Web socket seems blocked after sending more than 3 service requests

qiuzhong, 2018-03-14:
For the `call_service` operation, sending a request more than 3 times can cause a bridge client blocked. Here are the steps:

* Start the rosbridge server
```
$ node bin/rosbridge.js
The web socket server started on ws://localhost:9090
```

* Running the  client for 4 times:
```
'use strict';

/* eslint-disable camelcase */

const assert = require('assert');
const rclnodejs = require('rclnodejs');
const WebSocket = require('ws');

rclnodejs.init().then(() => {

  let node = rclnodejs.createNode('service');
  let wsService = node.createService('example_interfaces/srv/AddTwoInts', 'add_two_ints',
    (request, response) => {
      let result = response.template;
      result.sum = request.a + request.b;
      response.send(result);
      console.log('result sent');
    });
  rclnodejs.spin(node);

  const advertise_service = {op: 'advertise_service', type: 'example_interfaces/AddTwoInts', service: 'add_two_ints'};
  const call_service = {op: 'call_service', service: 'add_two_ints',
    args: {request: {a: 1, b: 2}, type: 'example_interfaces/AddTwoInts'}};

  let count = 0;
  let ws = new WebSocket('ws://localhost:9090');
  ws.on('open', function() {
    ws.send(JSON.stringify(call_service));
    count++;
  });
  ws.on('message', function(data) {
    let response = JSON.parse(data);
    console.log(`${count}`, response);
    count++;
    if (response.op == 'service_response') {
      ws.close();
      rclnodejs.shutdown();
    }
  });
}).catch((err) => {
  console.log(err);
});
```

We get this output for 1/2/3 time to run it.
```
1 { op: 'set_level', level: 'none' }
2 { op: 'service_response',
  service: 'add_two_ints',
  values: { sum: 3 },
  result: true }

```
However, for the 4th time, the client hangs and cannot receive the service response:
```
1 { op: 'set_level', level: 'none' }

```

minggangw, 2018-03-22:
I always got crash the second time I ran the test

```
 1: node::Abort() [node]
 2: 0x109f0ac [node]
 3: v8::Utils::ReportApiFailure(char const*, char const*) [node]
 4: node::TCPWrap::Instantiate(node::Environment*, node::AsyncWrap*) [node]
 5: node::ConnectionWrap<node::TCPWrap, uv_tcp_s>::OnConnection(uv_stream_s*, int) [node]
 6: 0x132baf0 [node]
 7: 0x1330ed8 [node]
 8: uv_run [node]
 9: node::Start(int, char**) [node]
10: __libc_start_main [/lib/x86_64-linux-gnu/libc.so.6]
11: 0x7b967d [node]
Aborted (core dumped)
```
Have you met this before @qiuzhong ?

qiuzhong, 2018-03-22:
Yes. This happened many times. But sometimes I encountered this as well: 
```
Segmentation fault (core dumped)
```

Without explicit error log.

minggangw, 2018-03-22:
You can execute to `gdb node -c core` debug it and get the same call stack as above.

minggangw, 2018-03-22:
Which version of node.js are you using? @qiuzhong

qiuzhong, 2018-03-23:
I tried with `Node.js  8.9.10`.

minggangw, 2018-03-23:
Hi @qiuzhong would you please install the `rclnodejs` module by:

```
npm install https://github.com/RobotWebTools/rclnodejs.git
```

I tested on it and found the case can pass, I suppose the problem is caused by https://github.com/RobotWebTools/rclnodejs/commit/5c2aa454492149b1b8ec250d6526e2d893aa2a84 which has already been fixed on the `develop` branch.

If the case can pass, I suggest publishing a bug fix release of `rclnodejs` on npmjs then. Thanks!

qiuzhong, 2018-03-23:
You mean you never encounter crash after you npm install with latest `rclnodejs`? I'll try it soon.

minggangw, 2018-03-23:
Yes, the crash never appeared again.

qiuzhong, 2018-03-23:
@minggangw , I uninstall `rclnodejs` and `npm install` with URL way. The crash still happened. However, for the first time after installation, it seems quite well to run the client above (It didn't crash for running the client more than 4 times and I almost thought it's fixed). But when I exit the terminal and open terminal/load ROS2 envionments/run the client, the crash came back again.

I wonder how do you get your ROS2 package? I built it from source before but not too old.

minggangw, 2018-03-23:
I build ROS from source, I think we'd better use the same [binary package](https://ci.ros2.org/view/packaging/job/packaging_linux/lastSuccessfulBuild/artifact/ws/ros2-package-linux-x86_64.tar.bz2) to test.

qiuzhong, 2018-03-23:
@minggangw , I tried the binary package. Crash still exists but there is some possibility that it does't crash for a second time to run the client, after that the crash never occurs.

minggangw, 2018-04-02:
rclnodejs [v0.3.5](https://www.npmjs.com/package/rclnodejs) has been published on npmjs, please verify this issue again, thanks!

qiuzhong, 2018-04-02:
The test case of this issue is used in PR#76, which is passed and this issue is fixed now.

qiuzhong, 2018-04-02:
Closing it.

closed 2018-04-02 by qiuzhong