RobotWebTools/ros2-web-bridge #70: Bridge server cannot be shutdown by CTRL + C at some condition

qiuzhong, 2018-03-14:
Fork from #69 , at some conditions, the bridge server cannot be shutdown by `CTRL + C`. You have to send `SIGTERM` signal to kill the bridge server.

[image]

As tin he screenshot, sending `CTRL + C` several times cannot shutdown the bridge server.

minggangw, 2018-03-22:
I will try to reproduce it and investigate, thanks!

minggangw, 2018-04-02:
rclnodejs [v0.3.5](https://www.npmjs.com/package/rclnodejs) has been published on npmjs, please verify this issue again, thanks!

qiuzhong, 2018-04-02:
This issue is not reproduced with the new release so far. Let's close it.

closed 2018-04-02 by qiuzhong