RobotWebTools/ros2-web-bridge #72: Both ros1 and ros2 message or interface types support

qiuzhong, 2018-03-16:
This proposal comes from the bridge client to interact with the `rclnodejs`.

In ROS1, like `roslibjs` the message or interface type is defined as `std_msgs/String`, `example_interfaces/AddTwoInts`.

In ROS2, like `rclnodejs`, the message or interface types is defined as `std_msgs/msg/String`, `example_interfaces/srv/AddTwoInts`.

Currently, the bridge server will assume all the message or interface types are in ROS1 format and insert `msg` or `srv`. When the service  is in `rclnodejs` environment, there might be inconsistent code.

Suggest to support both ROS1 and ROS2 message and interface types.

minggangw, 2018-03-22:
Do you mean the `ros2-web-bridge` can accept the type `example_interfaces/srv/AddTwoInts` and `example_interfaces/AddTwoInts` simultaneously?

qiuzhong, 2018-03-22:
Yes. This is more convenient.

minggangw, 2018-03-22:
Good idea, I will implement it.

PR #83 (2018-03-22): Support the format of message type for both ros1 and ros2
Currently when creating a publisher, we will pass something like
"std_msgs/String" as the type of the message. But the type of the message
is defined as "std_msgs/msg/String" in ros2.

This patch support both of these two formates.

Fix #72
commit 6a2a9dec (2018-03-22): Support the format of message type for both ros1 and ros2

Currently when creating a publisher, we will pass something like
"std_msgs/String" as the type of the message. But the type of the message
is defined as "std_msgs/msg/String" in ros2.

This patch support both of these two formates.

Fix #72
closed 2018-03-22 by minggangw