RobotWebTools/ros2-web-bridge #73: Implementation of ros2-api

vincentrou, 2018-03-16:
It would be interesting to have the [rosapi](https://github.com/RobotWebTools/rosbridge_suite/tree/develop/rosapi) in ros2.

But what is the best way to do it ?

- a dedicated python package/node that respond to services like in ros1
- a dedicated nodejs package/node that respond to services
- a direct integration in ros2-web-bridge, so it does not go through services; it ask directly to rclnodejs

Last two proposition needs to add the appropriate functions in rclnodejs to get topic and node list for example.

minggangw, 2018-03-22:
Hi @vincentrou 

Thanks for your suggestion. The clients designed for ROS2 are based on [rcl](https://github.com/ros2/rcl), so I think we'd better just bind the core client library other than introduce other functions in Node.js client. As you said, you can implement a simple ROS node as a service to offer what you want to grab :)

Currently we are implementing the [rosbridge v2 protocol](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md) based on `rclnodejs`. I think we will plan for the ROS2 tools and infrastructure in the next step.

FlorisE, 2020-08-28:
Hi @minggangw, thanks for replying to my question in the `rclnodejs` repo. I'm not sure if I understood you correctly, but do you think it would be suitable to add functionality similar to https://github.com/RobotWebTools/rosbridge_suite/tree/develop/rosapi to the `ros2-web-bridge`?

minggangw, 2020-08-28:
As the rosapi is not a standard in ROS2, I think it's up to the individual client to implement some facilities/utils, like rosapi. In ROS2, the `rosbridge` offers them. As a client of ROS2, `rclnodejs` follows some design principles https://index.ros.org/doc/ros2/Concepts/ROS-2-Client-Libraries/. But theoretically, a client could do more.

closed 2025-02-14 by vincentrou