RobotWebTools/ros2-web-bridge #78: service_response should be considered as a response operation

qiuzhong, 2018-03-20:
According to the [spec](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#2-the-rosbridge-protocol): `service_response` should be a response operation, which means it not considered as a request operation. Because it contains some necessary fields that cannot get from the request side.

```
{ "op": "service_response",
  (optional) "id": <string>,
  "service": <string>,
  (optional) "values": <list<json>>,
  "result": <boolean>
}
```
As a web socket client cannot get the result. Meanwhile, from the `call_service` will send a `service_response`, which proves the `service_response` is not a request operation.

However, current bridge implementation consider the `service_response` as a request operation:

```
    this._registerOpMap('service_response', (command) => {
      let serviceName = command.service;
      let id = command.id;
      let response = this._servicesResponse.get(id);
      if (response) {
        response.send(command.values);
        this._servicesResponse.delete(id);
      }
    });
```

minggangw, 2018-03-22:
If a service was created through `roslibjs` and when it received a request and sent the response back to the bridge, the `service_response` operation will be used, please see https://github.com/RobotWebTools/roslibjs/blob/3bc4ec2f08f2a1410e9af2959155f216943c790e/src/core/Service.js#L108

So the bridge may receive `service_response` and forward it to the client.

qiuzhong, 2018-03-22:
You're right. I was confused by the [spec](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#2-the-rosbridge-protocol):

> Response messages from the server are things that the client is giving back, so they are nouns (fragment, status, service_response etc.)

I checked the `service.html` and `client.html` and the `service_response` is considered as a request by `roslibjs` to send the result to the bridge. This procedure is automatically called by `roslibjs` and basically a third-party(like the UT) rarely uses this.

So I guess the spec is kind of confusin and should be corrected?

qiuzhong, 2018-03-22:
Or `service_response` can be both `request` and `response` operations?

minggangw, 2018-03-22:
I think the `Response messages` mentioned in the spec is a general conception, which means the response sent by the client.

`service_response` as an operation code can be generated by both `roslibjs` and the `bridge`, which is not clarified in the spec. I think the spec mingled these two things.

qiuzhong, 2018-03-22:
That's clear. Thanks! Close it.

closed 2018-03-22 by qiuzhong