RobotWebTools/ros2-web-bridge #93: set_level as a request operation

qiuzhong, 2018-04-13:
According to the [protocol spec](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#321-set-status-level--status_level--experimental), `set_level` is a response operation from the `ros2-web-bridge` to the `roslibjs`. 

However, there is a API in `roslibjs`: [Ros.setStatusLevel](https://github.com/RobotWebTools/roslibjs/blob/develop/src/core/Ros.js#L143)

```
Ros.prototype.setStatusLevel = function(level, id){
  var levelMsg = {
    op: 'set_level',
    level: level,
    id: id
  };

  this.callOnConnection(levelMsg);
};
```

This implies `set_level` should be a request from `roslibjs`. Currently, there is no handler for this kind of request operation.

minggangw, 2018-04-16:
In the  protocol spec [3.2](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#32-status-messages)

>rosbridge sends status messages to the client relating to the successes and failures of rosbridge protocol commands. There are four status levels: info, warning, error, none. By default, rosbridge uses a status level of error.

@qiuzhong have you find any code path which triggers the `setStatusLevel` function in roslibjs? This operation should not be executed by client side as I understand.

qiuzhong, 2018-04-16:
@minggangw , no, I didn't found any place that calling `Ros.setStatusLevel()` in roslibjs. It's just an exposed API. I thought if this API comply the spec, then bridge should support `set_level` as a request and vice versa.

So you think this API conflicts with the spec?

minggangw, 2018-04-18:
I have no idea why the `roslibjs` implemented this interface.