RobotWebTools/ros2-web-bridge #95: Cannot call a service with empty argument

vincentrou, 2018-04-18:
For now the service type is deduce from the args type.
But it should be possible to [call service with empty args](https://github.com/RobotWebTools/rosbridge_suite/blob/f0d359797eb90b322e5eb23cfda89ea7a9aceac8/ROSBRIDGE_PROTOCOL.md#346-call-service) 

To make it works, it needs to find the service type from the service name.

qiuzhong, 2018-04-19:
Please refer to #68 and this [PR](https://github.com/RobotWebTools/roslibjs/pull/282) in roslibjs

vincentrou, 2018-04-19:
Ok sorry I did not saw this PR and issue.
So you added the type field in the json. This will be an optional field ? What will you do if you have not the type, you will still look for the args ?
I was thinking that it should be doable by warping the [rcl_get_service_names_and_types](https://github.com/ros2/rclpy/blob/35275fac1e2136867549496de98060faee36a0a3/rclpy/src/rclpy/_rclpy.c#L2630) in rclnodejs.

qiuzhong, 2018-04-19:
@minggangw , would you please take a look.

minggangw, 2018-04-19:
@viktorku by calling the graph function [rcl_get_service_names_and_types](https://github.com/ros2/rcl/blob/d41c923927850274f63aee1bde565d3a8266a7bd/rcl/include/rcl/graph.h#L127), developer can get the detailed information of the designated `node`.

Here the bridge has a different scenario that the bridge itself can not obtain the `rcl_node_t` pointer e.g. a service offered by another node (I do not find a method of getting a service type by its name), therefore you have to pass the type explicitly in a string.

Correct me if I am wrong, thanks!

minggangw, 2018-05-18:
I'd like to close this issue as there is now update since last month.

closed 2018-05-18 by minggangw