RobotWebTools/ros2-web-bridge #98: Failed to receive complex type message via bridge

qiuzhong, 2018-05-30:
Let's say we send and receive complex type message like `sensor_msgs/Image`, here is the code based on 
[roslibjs]():

```
 var ros = new ROSLIB.Ros();

 ros.connect('ws://localhost:9090');
 var example = new ROSLIB.Topic({
     ros: ros,
     name: '/big_data_topic',
     messageType : 'sensor_msgs/Image'
  });
let dataLength = 320 * 240 * 4 * 4;
let uint8Data = new Array(dataLength);
for (let i = 0; i < dataLength; i++) {
  uint8Data[i] = i % 255;
}   
example.subscribe(function(message) {        
  var length = message.data.length;
  console.log(`${length} bytes received`);
});
      
example.publish({
  header: {stamp: {sec: 11223, nanosec: 44556}, frame_id: 'f001', },
  height: 240, width: 320, encoding: 'rgba', is_bigendian: false, step: 320*16, is_dense: false,
  data: Uint8Array.from(uint8Data),
});
```

Since there is no way to construct the message object and assign member for it, we can only publish message data in JavaScript object notation way, like above.

However, the subscriber failed to receive the message. We found the subscriber callback was never called and and got this error in from the `ros2-web-bridge` log:

```
  ros2-web-bridge:Bridge Exception caught in Bridge.executeCommand(): TypeError: msg.hasMember is not a function +1m
  ros2-web-bridge:Bridge Error: TypeError: msg.hasMember is not a function happened when executing command publish +1ms
  ros2-web-bridge:Bridge Response: {"op":"set_level","id":"publish:/big_data_topic:3","level":"error"} +0ms
```

The same JavaScript object notation way of message can be published and received by [rclnodejs node](https://github.com/RobotWebTools/rclnodejs/blob/develop/test/test-security-related.js#L259)

qiuzhong, 2018-05-30:
Move to `ros2-web-bridge` repo: [#98](https://github.com/RobotWebTools/ros2-web-bridge/issues/98)

minggangw, 2018-07-28:
We got this error because the JSON doesn't have a representation of  [typed array](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Typed_arrays), e.g.

``` JavaScript
let typedArray = new Int32Array([101, 102, 103]);
let str = JSON.stringify(typedArray);
// str is "{"0":101,"1":102,"2":103}"

let arr = JSON.parse(str);
// arr is Object {0: 101, 1: 102, 2: 103}
```

That is why we can't get the `typed array` back within the bridge.

minggangw, 2018-07-30:
One solution is that we can regress the typed array to the normal ones, so the array can be serialized to JSON correctly. In this way, we have to concern the inefficiency.

minggangw, 2018-08-15:
I have submitted https://github.com/RobotWebTools/rclnodejs/pull/409 to [rclnodejs](https://github.com/RobotWebTools/rclnodejs) to add an option to disable TypedArray. After it is landed, I will submit another PR to make the change in ros2-web-bridge accordingly.

minggangw, 2018-08-16:
Hi @qiuzhong Would you please test this issue base on #104, and assure to assign using **Array**, like:

```JavaScript
data: [1, 2, 3]
```

qiuzhong, 2018-08-16:
Sure. I'm testing this.

qiuzhong, 2018-08-16:
@minggangw , I tried the sample based on PR104 above. The data was assign with using **Array**. The error was gone! So I think it fixed the issue.

I wonder if we should never use the code of style like this:
` data: Uint8Array.from(uint8Data)`

minggangw, 2018-08-16:
>I wonder if we should never use the code of style like this:
data: Uint8Array.from(uint8Data)

If you use roslibjs to invoke the `rclnodejs`, you have to abandon the code like that.

PR #104 (2018-08-16): Disable the usage of TypedArray in rclnodejs
Because the JSON does not have a representation of TypedArray, we disable it
explicitly when subscription/client/service are being created.

Fix #98
commit 079f41a1 (2018-08-16): Disable the usage of TypedArray in rclnodejs

Because the JSON does not have a representation of TypedArray, we disable it
explicitly when subscription/client/service are being created.

Fix #98
closed 2018-08-16 by minggangw