RobotWebTools/ros2-web-bridge #106: “npm install” error

lxbeyond, 2018-08-21:
I git clone this rep, and I run "npm install", then get the error:
npm install

> ref@1.3.5 install /home/hzh/xinliu/ros2-web-bridge/node_modules/ref
> node-gyp rebuild

make: Entering directory '/home/hzh/xinliu/ros2-web-bridge/node_modules/ref/build'
  CXX(target) Release/obj.target/binding/src/binding.o
../src/binding.cc: In function ‘void init(Nan::ADDON_REGISTER_FUNCTION_ARGS_TYPE)’:
../src/binding.cc:643:8: warning: ‘Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’ is deprecated [-Wdeprecated-declarations]
   Nan::ForceSet(target, Nan::New<v8::String>("endianness").ToLocalChecked(), Nan::New<v8::String>(CheckEndianness()).
        ^
In file included from ../../nan/nan.h:197:0,
                 from ../src/binding.cc:7:
../../nan/nan_maybe_43_inl.h:130:35: note: declared here
 NAN_DEPRECATED inline Maybe<bool> ForceSet(
                                   ^
../src/binding.cc:643:8: warning: ‘Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’ is deprecated [-Wdeprecated-declarations]
   Nan::ForceSet(target, Nan::New<v8::String>("endianness").ToLocalChecked(), Nan::New<v8::String>(CheckEndianness()).
        ^
In file included from ../../nan/nan.h:197:0,
                 from ../src/binding.cc:7:
../../nan/nan_maybe_43_inl.h:130:35: note: declared here
 NAN_DEPRECATED inline Maybe<bool> ForceSet(
                                   ^
../src/binding.cc:643:187: warning: ‘Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’ is deprecated [-Wdeprecated-declarations]
 ed(), Nan::New<v8::String>(CheckEndianness()).ToLocalChecked(), static_cast<PropertyAttribute>(ReadOnly|DontDelete));
                                                                                                                    ^
In file included from ../../nan/nan.h:197:0,
                 from ../src/binding.cc:7:
../../nan/nan_maybe_43_inl.h:130:35: note: declared here
 NAN_DEPRECATED inline Maybe<bool> ForceSet(
                                   ^
../src/binding.cc:644:8: warning: ‘Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’ is deprecated [-Wdeprecated-declarations]
   Nan::ForceSet(target, Nan::New<v8::String>("NULL").ToLocalChecked(), WrapNullPointer(), static_cast<PropertyAttribu
        ^
In file included from ../../nan/nan.h:197:0,
                 from ../src/binding.cc:7:
../../nan/nan_maybe_43_inl.h:130:35: note: declared here
 NAN_DEPRECATED inline Maybe<bool> ForceSet(
                                   ^
../src/binding.cc:644:8: warning: ‘Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’ is deprecated [-Wdeprecated-declarations]
   Nan::ForceSet(target, Nan::New<v8::String>("NULL").ToLocalChecked(), WrapNullPointer(), static_cast<PropertyAttribu
        ^
In file included from ../../nan/nan.h:197:0,
                 from ../src/binding.cc:7:
../../nan/nan_maybe_43_inl.h:130:35: note: declared here
 NAN_DEPRECATED inline Maybe<bool> ForceSet(
                                   ^
../src/binding.cc:644:142: warning: ‘Nan::Maybe<bool> Nan::ForceSet(v8::Local<v8::Object>, v8::Local<v8::Value>, v8::Local<v8::Value>, v8::PropertyAttribute)’ is deprecated [-Wdeprecated-declarations]
 n::New<v8::String>("NULL").ToLocalChecked(), WrapNullPointer(), static_cast<PropertyAttribute>(ReadOnly|DontDelete));
                                                                                                                    ^
In file included from ../../nan/nan.h:197:0,
                 from ../src/binding.cc:7:
../../nan/nan_maybe_43_inl.h:130:35: note: declared here
 NAN_DEPRECATED inline Maybe<bool> ForceSet(
                                   ^
  SOLINK_MODULE(target) Release/obj.target/binding.node
  COPY Release/binding.node
make: Leaving directory '/home/hzh/xinliu/ros2-web-bridge/node_modules/ref/build'

> puppeteer@1.7.0 install /home/hzh/xinliu/ros2-web-bridge/node_modules/puppeteer
> node install.js

Downloading Chromium r579032 - 102.3 Mb [====================] 100% 0.0s 
Chromium downloaded to /home/hzh/xinliu/ros2-web-bridge/node_modules/puppeteer/.local-chromium/linux-579032

> rclnodejs@0.4.2 install /home/hzh/xinliu/ros2-web-bridge/node_modules/rclnodejs
> node-gyp rebuild

[eval]:1
console.log(process.env.COLCON_PREFIX_PATH.replace(/:/, '/include/ ') + '/include/')
                                           ^

TypeError: Cannot read property 'replace' of undefined
    at [eval]:1:44
    at ContextifyScript.Script.runInThisContext (vm.js:50:33)
    at Object.runInThisContext (vm.js:139:38)
    at Object.<anonymous> ([eval]-wrapper:6:22)
    at Module._compile (module.js:652:30)
    at evalScript (bootstrap_node.js:466:27)
    at startup (bootstrap_node.js:167:9)
    at bootstrap_node.js:612:3
gyp: Call to 'node -e "console.log(process.env.COLCON_PREFIX_PATH.replace(/:/, '/include/ ') + '/include/')"' returned exit status 1 while in binding.gyp. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:336:16)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
gyp ERR! System Linux 4.15.0-30-generic
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/hzh/xinliu/ros2-web-bridge/node_modules/rclnodejs
gyp ERR! node -v v8.11.2
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! rclnodejs@0.4.2 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the rclnodejs@0.4.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/hzh/.npm/_logs/2018-08-21T11_36_13_719Z-debug.log

and the node version is v8.11.2 , nvm version is : 0.33.11

qiuzhong, 2018-08-22:
This error occurred because the `COLCON_PREFIX_PATH` environment variable is not set while install the `rclnodejs` dependency.  Please check if had load the ROS2 environment? Please check your it by: `$ echo $AMENT_PREFIX_PATH`.

qiuzhong, 2018-08-22:
This is a frequent [error](https://github.com/RobotWebTools/rclnodejs/issues/396) recently. This root cause is RCL native APIs changed after ROS2 bouncy release and `rclnodejs` always keeps aligned with the latest RCL native APIs. This is a little annoying. You can:

* Use old version of `rclnodejs` that works with ROS2 bouncy, we don't recommend as there are bugs not fixed in old version.
* Use latest ROS2 package from CI, here is what we use in `rclnodejs` CI building:
  `wget https://ci.ros2.org/view/packaging/job/packaging_xenial_linux/lastSuccessfulBuild/artifact/ws/ros2-package-linux-x86_64.tar.bz2`

minggangw, 2018-08-22:
@lxbeyond If you have to build against bouncy release, I recommend that you install the `rclnodejs` from Github which locates at https://github.com/RobotWebTools/rclnodejs/tree/bouncy-bolson, it should work well. Thanks!

lxbeyond, 2018-08-22:
I build rclnodejs well from github bouncy-bolson branch, but error when build ros2-web-bridge , and I do not konw how to make ros2-web-bridge to use rclnodejs of bouncy-bolson version . @minggangw

minggangw, 2018-08-22:
You could change your [package.json](https://github.com/RobotWebTools/ros2-web-bridge/blob/develop/package.json#L46)

```
"rclnodejs": "^0.4.0",
```

please reference the [document](https://docs.npmjs.com/files/package.json#github-urls) of npm to change it to Github url, hope it works.

lxbeyond, 2018-08-22:
I changed package.json 
"rclnodejs": "git://github.com/RobotWebTools/rclnodejs.git#bouncy-bolson",
and it works, thanks @minggangw  @qiuzhong

minggangw, 2018-08-22:
So I think this issue can be closed now, feel free to reopen if you meet any problem in the future. Thanks!

closed 2018-08-22 by minggangw