RobotWebTools/ros2-web-bridge #108: run example "subscription.html" error

lxbeyond, 2018-09-13:
I test example as READ.me, run the command as follow:
      DEBUG=ros2-web-bridge:* node bin/rosbridge.js
      node ./index.js
and open "http://localhost:3000/html/subscription.html" ,then I get  following ouput:
`
ros2-web-bridge:index The ros2-web-bridge has started. +0ms
The web socket server started on ws://localhost:9090
  ros2-web-bridge:Bridge Web bridge 7b683e66-89d1-45ed-a537-f404790d3256 is created +0ms
  ros2-web-bridge:Bridge JSON command received: {"op":"subscribe","id":"subscribe:/example_topic:1","type":"std_msgs/String","topic":"/example_topic","compression":"none","throttle_rate":0,"queue_length":0} +1ms
  ros2-web-bridge:Bridge subscribe a topic named /example_topic +0ms
  ros2-web-bridge:Bridge Exception caught in Bridge.executeCommand(): TypeError: Invalid argument +16ms
  ros2-web-bridge:Bridge Error: TypeError: Invalid argument happened when executing command subscribe +0ms
  ros2-web-bridge:Bridge Response: {"op":"set_level","id":"subscribe:/example_topic:1","level":"error"} +1ms
`
I wonder what's the matter ?
@minggangw

qiuzhong, 2018-09-14:
Just tried the example with a recent ros2 package from CI and I couldn't reproduce this error.

@lxbeyond , what ros2 release package did you use?

lxbeyond, 2018-09-14:
I apply this demo in ros2 bouncy

lxbeyond, 2018-09-14:
and I run demos in rclnodejs in ros2 bouncy well

qiuzhong, 2018-09-14:
I failed to install ros2-web-bridge with the ros2 bouncy release. Would you please try to apply the ros2 package from [CI](https://ci.ros2.org/view/packaging/job/packaging_xenial_linux/lastSuccessfulBuild/artifact/ws/ros2-package-linux-x86_64.tar.bz2)

lxbeyond, 2018-09-14:
to install ros2-web-bridge with the ros2 bouncy release, you should change the package.json
"rclnodejs": "git://github.com/RobotWebTools/rclnodejs.git#bouncy-bolson"

lxbeyond, 2018-09-14:
I tride publisher.html example, and it worked.

lxbeyond, 2018-09-14:
I locate the error, and found it works when change the line https://github.com/RobotWebTools/ros2-web-bridge/blob/fa57972b5f4af998dc5b737094db34a84599b50c/lib/subscription_manager.js#L57 : 
from 
`
 let subscription = this._node.createSubscription(messageType, topicName, {enableTypedArray: false}, (message) => {
        this._subscripions.get(topicName).callbacks.forEach(callback => {
          callback(topicName, message);
        });
});
`
to
`
 let subscription = this._node.createSubscription(messageType, topicName, (message) => {
        this._subscripions.get(topicName).callbacks.forEach(callback => {
          callback(topicName, message);
        });
});
`

minggangw, 2018-09-14:
Would you please check the version of your `rclnodejs` used under `node_modules` folder?

lxbeyond, 2018-09-14:
I checked the version , the rclnodejs used under node_modules folder is the same to git://github.com/RobotWebTools/rclnodejs.git#bouncy-bolson

minggangw, 2018-09-14:
The `ros2-web-bridge` depends on `rclnodejs`, but we only adapt the `develop` branch other than `bouncy-bolson`.

So the ros2-web-bridge `develop` may be incompatible with non-`develop` branch of rclnodejs, as you have met.

minggangw, 2018-09-19:
I am going to close this issue, as it was caused by the incompatibility of using a wrong branch of `rclnodejs` with `ros2-web-bridge`. Feel free to reopen it if any other problem.

closed 2018-09-19 by minggangw