RobotWebTools/ros2-web-bridge #109: Header members incorrespondence

qiuzhong, 2018-09-14:
In roslibjs, we found the `builtin_interfaces/Time` members are `secs` and `nsecs`. While in ROS2, the members are `sec` and `nanosec`. When we work the bridge with roslibjs, there will be error. Which side should be updated to keep correspondence?

qiuzhong, 2018-09-14:
@minggangw

minggangw, 2018-09-14:
We must follow the definition of ROS 2 for the [Time](https://github.com/ros2/rcl_interfaces/blob/master/builtin_interfaces/msg/Time.msg) in order to use the right JavaScript message.

qiuzhong, 2018-09-14:
Should I file an issue for this in roslibjs repo?

minggangw, 2018-09-14:
I don't think this is a bug for roslibjs, as it origins from the change of .msg definition.

minggangw, 2018-09-19:
As this issue is caused by the different definition of ROS 2 from ROS 1,  I am going to close it, thanks!

closed 2018-09-19 by minggangw