RobotWebTools/ros2-web-bridge #113: npm i in docker says: cannot convert ‘rcutils_allocator_t’ to ‘rcl_context_t*’

azazdeaz, 2019-03-02:
Hi, i get the following error when i try to npm install in the [official ros2 docker image](https://hub.docker.com/r/osrf/ros2/):
```
make: Entering directory '/ros2-web-bridge/node_modules/rclnodejs/build'
  CXX(target) Release/obj.target/rclnodejs/src/addon.o
  CXX(target) Release/obj.target/rclnodejs/src/executor.o
../src/executor.cpp: In static member function ‘static void rclnodejs::Executor::Run(void*)’:
../src/executor.cpp:96:66: error: cannot convert ‘rcutils_allocator_t’ to ‘rcl_context_t*’ for argument ‘7’ to ‘rcl_ret_t rcl_wait_set_init(rcl_wait_set_t*, size_t, size_t, size_t, size_t, size_t, rcl_context_t*, rcl_allocator_t)’
                                       rcl_get_default_allocator());
```
logs with [node 8](https://pastecode.xyz/view/9cd62415) and [node 11](https://pastecode.xyz/view/8cfdbb22)

rclnodejs@0.9.0 seems to [install fine](https://pastecode.xyz/view/f13146b5)

ros2-web-bridge version: 0.2.5

How should i continue debugging this?

robotpilot, 2019-03-20:
Hi @minggangw ,
I have the same error. Do you know how to fix it?

minggangw, 2019-03-21:
Sorry for the late response @azazdeaz @robotpilot 

I checked the error, it seems that it's the incompatibility problem caused by the rcl interface change. So would you please check the release of ROS 2.0 you used in the docker image, and you can find the current official releases at https://github.com/ros2/ros2/releases.

The 0.9.0 should work with the latest crystal patch release3. Through your error log, I suspect that the docker image is base on the master branch of [ROS 2.0](https://github.com/ros2/ros2). If it does, please use the master branch of rclnodejs by changing it in your package.json.

If the error still happens, please tell me and I will double check. Thanks!

azazdeaz, 2019-03-30:
Thanks! Unfortunately, i get the same error with a newly built image. (I couldn't figure out how to tell the release number from the downloaded binaries ( ? :) ) but it should pull the latest during docker build. )

minggangw, 2019-04-01:
I see. I suggest you install `rclnodejs` from github directly, e.g. open the `package.json` of your app, and add the line below to replace the original dependency:

```"rclnodejs": "RobotWebTools/rclnodejs"```

This should curb the compiling error.

azazdeaz, 2019-04-13:
Thank you! The install finishes without error with
ros2-web-bridge - https://github.com/RobotWebTools/ros2-web-bridge/commit/2fef243963ee1ad7a0a0568eda2d430f8729f5d1
rclnodejs - https://github.com/RobotWebTools/rclnodejs/commit/d63edca186997e6c682fc690218ab3f06b245a30
ros2 image - [april11 nightly build](https://hub.docker.com/r/osrf/ros2/tags)

closed 2019-04-13 by azazdeaz