RobotWebTools/ros2-web-bridge #117: ros2-web-bridge and rosbridge_suite

dejanpan, 2019-04-18:
I learnt today that we have 2 similar projects going: 
1. https://github.com/RobotWebTools/ros2-web-bridge
1. https://github.com/RobotWebTools/rosbridge_suite

On the high level they both seem to be doing the same thing: support websockets and bidirectionally communicate with ROS. The only exception that ros2-web-bridge works for ROS 2 and rosbridge_suite (currently) for ROS 1. 

Is this correct? If yes, do you plan to merge both projects at some point?

dejanpan, 2019-04-18:
@minggangw

minggangw, 2019-04-19:
Yes, the functions are the same, except `ros2-web-bridge` is designed for ROS2. As for merging them into one project, well I think it's impossible because the backends (ROS1 is python implementation, ROS2 is JavaScript implementation) for them are different.

The reason why they can somehow do the same thing is that they follow the same [protocol](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md) which is support by the library [roslibjs](https://github.com/RobotWebTools/roslibjs) to communicate with the browser over websockets.