RobotWebTools/ros2-web-bridge #118: Can i use npm install ros2-web-bridge ro install it?

shmily-skx, 2019-04-20:
I have already installed ROS2 and need to use ros2-web-bridge, but I don't know how to use it. After installing nodejs and npm, enter the command npm install. Then move to the ros2-web-bridge folder under clone and run the node bin/rosbridge.js command? But I am prompted to have an error.

minggangw, 2019-04-20:
Thanks for your question. I think about some steps you should check:

- Have you source the `.bash` file before starting (please reference installation of ROS 2 at [here](https://index.ros.org/doc/ros2/Installation/))
- If you want to install by `npm`, please run `npm i ros2-web-bridge` under the folder of your own app or add it as dependency in your `package.json` file.
- Finally, you can follow the [example](https://github.com/RobotWebTools/ros2-web-bridge#run-examples) to have a try, it should work.

Good luck!

shmily-skx, 2019-04-20:
> Thanks for your question. I think about some steps you should check:
>
>     * Have you source the `.bash` file before starting (please reference installation of ROS 2 at [...]

I have the following problem
[image]

minggangw, 2019-04-20:
Please follow the steps https://index.ros.org/doc/ros2/Installation/Linux-Install-Binary/#try-some-examples to see if the ros2 example can work properly.

Then run `echo $AMENT_PREFIX_PATH` to check if the env variable is set.

rukie, 2019-04-26:
I also get this error. 
From the log file:
```
8462 verbose stack Error: ref@1.3.5 install: `node-gyp rebuild`
```
I'm running a clean 18.04 install with ROS2 Crystal. I've verified the talker and listener example scripts run.

minggangw, 2019-04-29:
Hi @rukie would you please check the nodejs version you installed, you can reference below:

```bash
nvm install v10.15.2
```

rukie, 2019-04-29:
That worked! Thank you. 
This would be the complete documentation of my install, if it helps anyone:

Install NVM per their site's instructions.
```
cd ~/ 
git clone https://github.com/nvm-sh/nvm.git .nvm
cd ~/.nvm 
git checkout v0.34.0
source ~/.nvm/nvm.sh
```
Add these lines to your ~/.bashrc

```
export NVM_DIR="$HOME/.nvm"
[ -s "$NVM_DIR/nvm.sh" ] && \. "$NVM_DIR/nvm.sh"  # This loads nvm
[ -s "$NVM_DIR/bash_completion" ] && \. "$NVM_DIR/bash_completion"  # This loads nvm bash_completion
```

Install node.js with:
```
nvm install node
```

Clone ROS2-web-bridge
```
Git clone https://github.com/RobotWebTools/ros2-web-bridge
Cd ros2-web-bridge
```

Check version and install ROS2-Web-Bridge
```
nvm install v10.15.2
Npm install
```

closed 2019-08-12 by minggangw