RobotWebTools/ros2-web-bridge #119: Ubuntu NodeJS + npm

rukie, 2019-04-26:
Is there any reason you suggest installing NodeJS and npm from sources, rather than what's available through Ubuntu's package repository? 

I would think this would be easier for more users. 

```
sudo apt install nodejs npm
```

rukie, 2019-04-26:
Although it seems like that conflicts with a ros-crystal installation.... kinda defeating the purpose there huh.

minggangw, 2019-04-29:
Sorry for the late response, the reason why we suggest to install the nodejs through [nvm](https://github.com/nvm-sh/nvm) or download the latest LTS version is that this bridge leverages [rclnodejs](https://github.com/RobotWebTools/rclnodejs) as a backend to access the ROS2, and rclnodejs is actually an C++ addon which does the binding of C library (here is [rcl](https://github.com/ros2/rcl)) to JavaScript.

But there are some compatibility problems if you use different version of nodejs (due to the change of v8), currently we are testing the project on the latest LTS of nodejs, which offers better support of ES6 features.

rukie, 2019-04-29:
Thanks for the background!

closed 2019-04-29 by rukie