RobotWebTools/ros2-web-bridge #120: Error npm install

ErijBNT, 2019-05-29:
I am  always getting this error while running npm install 
[image]

minggangw, 2019-08-12:
The latest version of ros2-web-bridge ([v0.2.7](https://www.npmjs.com/package/ros2-web-bridge/v/0.2.7)) has been released, which was verified on [travis-ci](https://travis-ci.org/RobotWebTools/ros2-web-bridge/builds/570655926?utm_source=github_status&utm_medium=notification) with [dashing patch release2](https://github.com/ros2/ros2/releases/tag/release-dashing-20190806). You can reference the configuration of the CI to setup your own env, thanks!

minggangw, 2020-01-17:
I think it doesn't happen any more, so let's close it.

closed 2020-01-17 by minggangw