RobotWebTools/ros2-web-bridge #124: npm install "rcl/error_handling.h: No such file or directory"

salsicha, 2019-08-02:
ROS: ros-dashing
ROS2-web-bridge: latest 'develop' branch

"npm install" produces:
...
make: Entering directory '/home/up/repos/ros2-web-bridge/node_modules/rclnodejs/build'
  CXX(target) Release/obj.target/rclnodejs/src/addon.o
  CXX(target) Release/obj.target/rclnodejs/src/executor.o
../src/executor.cpp:17:10: fatal error: rcl/error_handling.h: No such file or directory
 #include <rcl/error_handling.h>
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
rclnodejs.target.mk:126: recipe for target 'Release/obj.target/rclnodejs/src/executor.o' failed
make: *** [Release/obj.target/rclnodejs/src/executor.o] Error 1
make: Leaving directory '/home/up/repos/ros2-web-bridge/node_modules/rclnodejs/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:198:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Linux 5.0.0-23-generic
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/up/repos/ros2-web-bridge/node_modules/rclnodejs
gyp ERR! node -v v10.16.0
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! rclnodejs@0.3.5 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the rclnodejs@0.3.5 install script.
...

I tried changing the package.json "rclnodejs" line to "git://github.com/RobotWebTools/rclnodejs.git", "0.9.0", and "git://github.com/RobotWebTools/rclnodejs.git#bouncy-bolson" but none of them worked....

minggangw, 2019-08-11:
Please double check with ```echo $AMENT_PREFIX_PATH``` to ensure that the include path has been added, thanks.

minggangw, 2019-08-12:
The latest version of ros2-web-bridge ([v0.2.7](https://www.npmjs.com/package/ros2-web-bridge/v/0.2.7)) has been released, which was verified on [travis-ci](https://travis-ci.org/RobotWebTools/ros2-web-bridge/builds/570655926?utm_source=github_status&utm_medium=notification) with [dashing patch release2](https://github.com/ros2/ros2/releases/tag/release-dashing-20190806). You can reference the configuration of the CI to setup your own env, thanks!

ryo4432, 2019-09-23:
I got the similar error and solve it.

Ubuntu: 18.04
ROS: ros-dashing
ros2-web-bridge: v0.2.7

In my case, before `npm install`, I removed the below script in my .bashrc and reopened the terminal in order to clear the `$COLCON_PREFIX_PATH`.
```
source path/to/ros2_repo/install/local_setup.bash
```
(`ros2_repo` is my local ros2 repository folder.)

I think that `$AMENT_PREFIX_PATH` is recognized as the include path when `$COLCON_PREFIX_PATH` is registered nothing ( you can check it with `echo $COLCON_PREFIX_PATH`).

minggangw, 2019-09-23:
This has been fixed by https://github.com/RobotWebTools/rclnodejs/pull/508 on the latest rclnodejs [v0.10.2](https://github.com/RobotWebTools/rclnodejs/releases/tag/0.10.2), I am going to upgrade the rclnodjs later.

minggangw, 2020-01-17:
I think it doesn't happen any more, so let's close it.

closed 2020-01-17 by minggangw