RobotWebTools/ros2-web-bridge #128: Is there a "/connected_clients" topic?

ramezanifar, 2019-10-16:
Hi,
I have few questions:
Q1)
In rosbridge_suite (for ros 1), there are two very useful topics namely
/client_count
/connected_clients
I don't find these two in ros2-web-bridge.  Do we have them here?

Q2)
When a client connects to the web socket server, in ros 1 bridge there are  echo messages showing that a client has connected or disconnected and what topics it has subscribed to. I don't get these messages in ros2-web-bridge. Are there such messages here?

Q3)
I can have my desired port when I run the websocket server using the arguemnt
$node index.js -p 9010
but the IP address does not change for example:
$node index.js -a '192.168.1.100'
still runs the server on localhost.

Thank you in advance

minggangw, 2019-10-17:
Hi @ramezanifar, thanks for your questions!

> In rosbridge_suite (for ros 1), there are two very useful topics namely

Unfortunately, ` ros2-web-bridge` doesn't offer these two topics now.

>Are there such messages here?

If you want to see what's going on as echo message, you can enable [debug](https://www.npmjs.com/package/debug):
`DEBUG=ros2-web-bridge:*`

>$node index.js -a '192.168.1.100'

Currently, we leverage [ws](https://www.npmjs.com/package/ws) to set up a server, do you want to use an external server? like: https://github.com/websockets/ws#external-https-server

ramezanifar, 2019-10-18:
Thank you for the reply
> Unfortunately, ros2-web-bridge doesn't offer these two topics now.

Please consider them.

> DEBUG=ros2-web-bridge:* 

Where exactly do you put this line?  Is this a bash command?

Thank you

minggangw, 2019-10-21:
1. I'd like to add them if I have spare time :)
2. It's for the bash command, you can check out https://www.npmjs.com/package/debug for more information.