RobotWebTools/ros2-web-bridge #130: ROS2 Actions Support

mikelyndersOKCC, 2020-01-15:
I have a created a ROS2 package that implements ROS2 Actions and it would be great if I could call them through my webapp interface. Unfortunately, it seems ROS2 Actions are not supported yet.

What would it take to get this functionality? I have gathered that there are a couple reasons why this might be hard:

- [rclnodejs ](https://github.com/RobotWebTools/rclnodejs)does not support ROS2 actions
- [rosbridge v2.0 protocol](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md) does not address ROS2 actions 
- [roslibjs ](https://github.com/RobotWebTools/roslibjs/blob/develop/examples/fibonacci.html) seems to support actionlib ROS1 actions, but I don't know/understand if that functionality can be used with ROS2 actions

How could one assist in the effort to support ROS2 Actions in ROS2-web-bridge?

Would ROS2 actions need to be implemented inside roslibjs using the topics and services that are already supported by ROS2-web-bridge and the rosbridge v2.0 protocol? (I think this is more or less how roslibjs implemented the actionlib support)

or 

Should actions support be implemented in rclnodejs first and then rosbridge v2.0 protocol be extended to support actions in some way?

I'd love to help in any way I can. I'm fairly new to ROS so please correct me if I'm misunderstanding.

In the meantime I will likely change my ROS package to provide the same functionality with a long-running service. (I know that is bad as services are meant to 'return quickly' but I'm not sure how else to do it)

mikelyndersOKCC, 2020-01-16:
Thanks for the explanation! I understand much better, I will take a look at [RobotWebTools/rclnodejs#469](https://github.com/RobotWebTools/rclnodejs/issues/469) as well as the guidelines in  [RobotWebTools/rclnodejs#498](https://github.com/RobotWebTools/rclnodejs/issues/498) to determine if this is something I might be able to contribute to!

gautamjain, 2020-12-03:
Looks like actions are now supported in rclnodejs.  Would extending the rosbridge protocol be the next step?

On a related note - is the 'rosbridge v2.0 protocol' meant for ROS2?  Or is it the second iteration of a protocol designed for ROS1?

minggangw, 2020-12-05:
> Looks like actions are now supported in rclnodejs. Would extending the rosbridge protocol be the next step?

Yes, rclnodejs now supports actions, but we haven't had plan to extend the rosbridge protocol yet.

> On a related note - is the 'rosbridge v2.0 protocol' meant for ROS2? Or is it the second iteration of a protocol designed for ROS1?

It stands for the 2nd version of rosbridge protocol for ROS1, that's is why it's difficult to extend to ROS2.

spygibas, 2021-02-17:
Hi, I make website and use ros2-web-bridge to connect with ros2, I have problem on navigation2 because I can't use actions roslibjs and can't get result or feedback if I send pose direct via geometry_msgs/msg/PoseStamped.

Can anyone help me for solution for these problem ?
Would better way I should do ?
1. Maybe guideline how to me upgrade roslibjs or ros2-web-bridge to support action ros2.
or
2. I'm new about nodejs but I don't know rclnodejs  need to use with ros2-web-bridge if I need use on website for connect ros2 ex. website server monitor and control to robot.
or
3. Modify navigation2 (cpp)  to publish some topic after  robot navgation done for result.
4.  Other...

minggangw, 2021-02-17:
Currently,  `ros2-web-bridge` doesn't support the actions feature for ROS2, because the way of implementing action in ROS2 is different from ROS1 and `roslibjs` is designed for ROS1. The `rclnodejs` acts as the back-end for `ros2-web-bridge` and `ros2-web-bridge` acts as a kind of bridge to connect the web and the ROS2 nodejs client (rclnodejs). You could consider it as a proxy to send/receive messages to/from the web server.

spygibas, 2021-02-17:
> Currently, `ros2-web-bridge` doesn't support the actions feature for ROS2, because the way of implementing action in ROS2 is different from ROS1 and `roslibjs` is designed for ROS1. The `rclnodejs` [...]

Thank you for explain. I understand now between rclnodejs and ros2-web-bridge.

nadavis, 2023-06-16:
@spygibas did success to manage it? i have the same issue? any good advice?

spygibas, 2023-06-17:
> @spygibas did success to manage it? i have the same issue? any good advice?

rclnodejs acts as the back-end server at location robot , not same as roslibjs it at location on server website
if you need use need to make conversation between rclnodejs first

spygibas, 2023-06-21:
@nadavis  Hi, I suggest you to use these https://github.com/open-rmf/rmf-web
for demo test https://github.com/open-rmf/rmf_demos