RobotWebTools/ros2-web-bridge #143: npm install error ROS2 Dashing

repa1030, 2020-04-17:
Hello,

When I follow the install instructions, I get an build error of npm install under ROS2 Dashung, Ubuntu 18.04, current development Branch of Ros2-Web-Bridge.

```
rclnodejs.target.mk:134: recipe for target 'Release/obj.target/rclnodejs/src/rcl_bindings.o' failed
make: *** [Release/obj.target/rclnodejs/src/rcl_bindings.o] Error 1
make: Leaving directory '/home/ros/ros2-web-bridge/node_modules/rclnodejs/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:310:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
gyp ERR! System Linux 5.3.0-46-generic
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/ros/ros2-web-bridge/node_modules/rclnodejs
gyp ERR! node -v v12.16.2
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@~2.1.1 (node_modules/chokidar/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! rclnodejs@0.13.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the rclnodejs@0.13.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/ros/.npm/_logs/2020-04-17T21_14_17_721Z-debug.log

```

Thank for your help!

minggangw, 2020-04-19:
It seems that installing the [rclnodejs](https://github.com/RobotWebTools/rclnodejs) failed. rclnodejs is a Node.js client for ROS2 and the latest version is [0.13.0](https://www.npmjs.com/package/rclnodejs/v/0.13.0), which is compatible with [Eloquent Elusor Patch Release 1](https://github.com/ros2/ros2/releases/tag/release-eloquent-20200124). Because you are using Dashing, I think that causes the failure (an [issue](https://github.com/RobotWebTools/rclnodejs/issues/640) has been submitted when using rclnodejs v0.13.0 with Dashing). 

Till now, we don't have a specific version supporting Dashing after Patch release 4, I suggest you could upgrade your ROS2 to Eloquent Elusor if possible. Thanks!

repa1030, 2020-04-20:
Thank you, fixed!

closed 2020-04-20 by repa1030