RobotWebTools/ros2-web-bridge #144: Npm install error

ctgloglu, 2020-05-14:
Npm install output for windows 10
> TypeError: Cannot read property 'replace' of undefined
>     at [eval]:1:43
>     at Script.runInThisContext (vm.js:120:20)
>     at Object.runInThisContext (vm.js:311:38)
>     at Object.<anonymous> ([eval]-wrapper:10:26)
>     at Module._compile (internal/modules/cjs/loader.js:1158:30)
>     at evalScript (internal/process/execution.js:94:25)
>     at internal/main/eval_string.js:23:3
> gyp: Call to 'node -e "console.log(process.env.AMENT_PREFIX_PATH.replace(/;/g, '\\include ').replace(/\\/g, '/') + '/include')"' returned exit status 1 while in binding.gyp. while trying to load binding.gyp
> gyp ERR! configure error
> gyp ERR! stack Error: `gyp` failed with exit code: 1
> gyp ERR! stack     at ChildProcess.onCpExit (C:\Users\Can\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\configure.js:351:16)
> gyp ERR! stack     at ChildProcess.emit (events.js:311:20)
> gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
> gyp ERR! System Windows_NT 10.0.18363
> gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\Can\\AppData\\Roaming\\npm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
> gyp ERR! cwd C:\Users\Can\Desktop\ros2-web-bridge-develop\node_modules\rclnodejs
> gyp ERR! node -v v12.16.1
> gyp ERR! node-gyp -v v5.1.0
> gyp ERR! not ok
> npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@~2.1.1 (node_modules\chokidar\node_modules\fsevents):
> npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})
> 
> npm ERR! code ELIFECYCLE
> npm ERR! errno 1
> npm ERR! rclnodejs@0.14.0 install: `node-gyp rebuild`
> npm ERR! Exit status 1
> npm ERR!
> npm ERR! Failed at the rclnodejs@0.14.0 install script.
> npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
> 
> npm ERR! A complete log of this run can be found in:
> npm ERR!     C:\Users\Can\AppData\Roaming\npm-cache\_logs\2020-05-14T08_10_12_414Z-debug.log
>

CLansdaleBEC, 2020-05-28:
I am getting the same issue with Ubuntu 20.04, node 10.19

minggangw, 2020-05-28:
For linux, make sure you have ` source <path/to/ros2>/install/local_setup.bash`
For Windows, ` call <path\to\ros2>\install\local_setup.bat`

CLansdaleBEC, 2020-05-28:
Confirming I sourced the ROS2 setup file. As a workaround I dropped a working package-lock.json in  from and old build I had and was able to build with that successfully.

CLansdaleBEC, 2020-05-29:
I tried a fresh install on a VM running dashing on Ubuntu 18.04 and get the same issue. Considering the package-lock json seems to fix things could the current version be incompatible with rclnodejs?

minggangw, 2020-05-30:
Hmm, I checked the usage of `package-lock-json` and [found](https://docs.npmjs.com/configuring-npm/package-lock-json.html): 

> It describes the exact tree that was generated, such that subsequent installs are able to generate identical trees, regardless of intermediate dependency updates.

So I think the file itself will not affect the process of compiling of C++ code, maybe something else leads to the failure.

roddc, 2020-06-11:
Hey @minggangw, I got this error when installing web bridge in docker, I have sourced ROS2 before installation. 

../src/rcl_action_bindings.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE rclnodejs::ActionProcessCancelRequest(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/rcl_action_bindings.cpp:631:36: warning: ignoring return value of ‘rcl_ret_t rcl_action_cancel_response_fini(rcl_action_cancel_response_t*)’, declared with attribute warn_unused_result [-Wunused-result]
     rcl_action_cancel_response_fini(cancel_response_ptr);
     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
../src/rcl_bindings.cpp:33:10: fatal error: rosidl_runtime_c/string_functions.h: No such file or directory
 #include <rosidl_runtime_c/string_functions.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Release/obj.target/rclnodejs/src/rcl_bindings.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/opt/node-v12.16.1-linux-x64/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:311:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
gyp ERR! System Linux 5.3.0-53-generic
gyp ERR! command "/opt/node-v12.16.1-linux-x64/bin/node" "/opt/node-v12.16.1-linux-x64/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /opt/node-v12.16.1-linux-x64/lib/node_modules/ros2-web-bridge/node_modules/rclnodejs
gyp ERR! node -v v12.16.1
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! rclnodejs@0.14.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the rclnodejs@0.14.1 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

minggangw, 2020-06-11:
The rclnodejs [v0.14.1](https://www.npmjs.com/package/rclnodejs/v/0.14.1) supports the latest ROS2 [Foxy](https://github.com/ros2/ros2/releases/tag/release-foxy-20200605), but not for Eloquent Elusor because these two ROS2 releases are incompatible.

From the error message, I suppose you are not running Foxy. The solution is that you could pin a specific rclnodejs version in [package.json](https://github.com/RobotWebTools/ros2-web-bridge/blob/develop/package.json#L46), e.g.

```
"rclnodejs": "0.14.0",
```
If you want to run on another version of ROS2, please check out this [compatibility table](https://github.com/RobotWebTools/rclnodejs#match-with-ros-20-stable-releases) to select the right version explicitly, thanks!

minggangw, 2020-06-23:
I think we could close it now, please feel free to reopen if you still have problems, thanks!

closed 2020-06-23 by minggangw