RobotWebTools/ros2-web-bridge #145: Connected but no topics

maxlein, 2020-05-18:
Hi,

just tried your bridge and i am wondering how I can debug the bridge more.  
I only get:  

```
Websocket started on ws://localhost:9090
(node:38087) DeprecationWarning: Deep requiring like `const uuidv4 = require('uuid/v4');` is deprecated as of uuid@7.x. Please require the top-level module when using the Node.js CommonJS module or use ECMAScript Modules when bundling for the browser. See https://github.com/uuidjs/uuid#deep-requires-now-deprecated for more information.
```  

The warning comes when I connect to the bridge.  
I was using the ros1 bridge before with the same data and I could echo my data there.  
With ros1 bridge there is log output when new topics are registered.  
Is there something similar with this lib?

minggangw, 2020-05-19:
Thanks for your question and hope I can help, see below:

> The warning comes when I connect to the bridge.

This warning will not influence the functionalities I think.

> Is there something similar with this lib?

Please reference https://github.com/RobotWebTools/ros2-web-bridge/issues/33#issuecomment-362468184 which may help you, thanks!

maxlein, 2020-05-19:
Thanks, now I am getting some Info!   

So my JSON looks right, but I don't know where this invalid argument "sec" comes from!?  
Should it maybe be "secs"?

```
ros2-web-bridge:Bridge JSON command received: {"topic":"unity_pose","msg":{"header":{"seq":1099,"stamp":{"secs":23,"nsecs":609434127},"frame_id":"base_link"},"pose":{"position":{"x":-0.00117289787,"y":-0.7621595,"z":-0.182284892},"orientation":{"x":0.000002394576,"y":0.000699281169,"z":-0.000125215534,"w":0.999999762}}},"op":"publish","id":"unity_pose"} +0ms
  ros2-web-bridge:Bridge Publish a topic named unity_pose with {"header":{"seq":1099,"stamp":{"secs":23,"nsecs":609434127},"frame_id":"base_link"},"pose":{"position":{"x":-0.00117289787,"y":-0.7621595,"z":-0.182284892},"orientation":{"x":0.000002394576,"y":0.000699281169,"z":-0.000125215534,"w":0.999999762}}} +0ms
  ros2-web-bridge:Bridge Response: {"op":"status","level":"error","msg":"publish: TypeError: Invalid argument: sec in Time","id":"unity_pose"} +1ms
```

minggangw, 2020-05-20:
Yes, the definition of `Time` is different in ROS1 and ROS2, see:

ROS1: http://wiki.ros.org/msg
ROS2: https://github.com/ros2/rcl_interfaces/blob/master/builtin_interfaces/msg/Time.msg

I think you have to rename the keys if you want to publish the topic to ROS2.

closed 2020-05-20 by maxlein

newcanopies, 2021-04-23:
hi @maxlein @minggangw 

Just ran into the same issue in `ros2-web-bridge` with Message time definition discrepancy
`ros2-web-bridge:Bridge Response: {"op":"status","level":"error","msg":"publish: TypeError: Invalid argument: sec in Time","id":"publish:/move_base_simple/goal:2"} +4ms`

If I rename the keys in the [Time.msg](https://github.com/ros2/rcl_interfaces/blob/master/builtin_interfaces/msg/Time.msg) as suggested by @minggangw  from secs to sec, 
how to make this custom message definition available to `ros2-web-bridge` ?

Is it possible to avoid creating an entire custom-message, to ensure all the functionalities of the base message type?

newcanopies, 2021-04-23:
hi   @maxlein @minggangw

Just ran into the `ros2-web-bridge` Error response to the Time message definition discrepancy 
```
ros2-web-bridge:Bridge   Response:   {"op":"status","level":"error","msg":"publish:   TypeError: Invalid argument: sec in   Time","id":"publish:/move_base_simple/goal:2"} +4ms
```
 
**What is the source of discrepancy in this environment?** Is it the client-side `roslibjs`?

ROS2  (Time definition _secs_)
Gazebo11
`ros2-web-bridge`
rclnodejs 0.17.1
node 12.20.1

Message Object
```
        var PoseStamped = new ROSLIB.Message({
            header: {
              seq: 0,
              stamp: {
                  secs: 0,
                  nsecs: 100
              },
              frame_id: "map"
            },
            pose: {
                position: {
                    x: 0.0,
                    y: 0.0,
                    z: 0.0
                },
                orientation: {
                    x: 0.0,
                    y: 0.0,
                    z: 0.0,
                    w: 1.0
                }
            }
        });
```