RobotWebTools/ros2-web-bridge #147: Ros2 Launch file

MRedel21, 2020-06-05:
Is it possible to start the package via the ros2 launchfile.py?

minggangw, 2020-06-05:
Currently, this module, ros2-web-bridge, has not been integrated with [ros2cli](https://github.com/ros2/ros2cli)  if that's what you want to have.

mis-eu, 2020-10-08:
would be great to have the feature of launching with ros2 launchfile.py on 20.04/Foxy

minggangw, 2020-10-08:
We need to change ros2cli to support this feature, so we don't further plan right now.

spygibas, 2021-02-18:
I think you can do via msg std_msgs/msg for make execute a launch/kill files example on these link below but you need to update that code for support some package for ros2 
https://answers.ros.org/question/268044/can-i-execute-a-launch-file-or-run-a-node-through-roswebtool/

closed 2021-03-19 by M-Redel