RobotWebTools/ros2-web-bridge #149: Header Timestamp automatically

MRedel21, 2020-06-25:
hi guys, if I omit the header timestamp, according to this description it should read: "Alternatively, just the timestamp field can be omitted, and then the current time will be automatically inserted. "But if I omit the timestamp then I won't get any messages in ROS2.

minggangw, 2020-06-25:
You could enable the debug to see the detailed information, e.g.

```
DEBUG=ros2-web-bridge* node bin/rosbridge.js
```

MRedel21, 2020-06-29:
Hello, I have bent it and I have received the following message. But the question was if I can omit the timestamp in the json transfer and have it added automatically when the message is received in ROS2.  
This worked in ROS 1. 
[image]

minggangw, 2020-06-29:
> But the question was if I can omit the timestamp in the json transfer and have it added automatically when the message is received in ROS2.

Currently, the `rclnodejs`, as the back-end of the bridge, will not assign a default value to a key (e.g. `timestamp`) if it's not been explicitly assigned to. Further, the `rclnodejs` will validate the message/topic when it's being published, so an error will be reported which I think it's what you meet.

closed 2021-03-19 by M-Redel