RobotWebTools/ros2-web-bridge #150: Failed at the rclnodejs@0.15.1 install script

MarcoAndrade26, 2020-08-04:
Hi,
I am trying to install ros2-web-bridge using npm i ros2-web-bridge.

I am in Ubuntu 18.04 and ROS2 Dashing.

Node version: 10.22.0
Npm version: 6.14.6
Rclnodejs version: 0.10.3

How can I install ros2-web-bridge for rclnodejs 0.10.3?

minggangw, 2020-08-05:
Please reference https://github.com/RobotWebTools/ros2-web-bridge/issues/144#issuecomment-642373204

MarcoAndrade26, 2020-08-06:
> Please reference [#144 (comment)](https://github.com/RobotWebTools/ros2-web-bridge/issues/144#issuecomment-642373204)

Thanks! it works! Is there any way to execute actions using ros2-web-bridge?

minggangw, 2020-08-07:
Currently, `ros2-web-bridge` doesn't support actions feature, and if you want to run your apps in browser, I suggest you could try [Electron](https://www.electronjs.org/) which supports Node.js env by using [rclnodejs](https://github.com/RobotWebTools/rclnodejs) directly (rclnodejs is a full-feature Node.js client of ROS2).

minggangw, 2020-08-10:
Close this issue as fixed.

closed 2020-08-10 by minggangw