RobotWebTools/ros2-web-bridge #151: Problem when installing in Dockerfile

maxlein, 2020-08-10:
I am trying to create a Dockerfile for easy startup of the bridge and I encountered this error during call to `npm install`:  

```
rclnodejs@0.15.2 install /root/ros2-web-bridge/node_modules/rclnodejs
> node-gyp rebuild

[eval]:1
console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/ ') + '/include/')
                                          ^

TypeError: Cannot read property 'replace' of undefined
    at [eval]:1:43
    at Script.runInThisContext (vm.js:131:20)
    at Object.runInThisContext (vm.js:297:38)
    at Object.<anonymous> ([eval]-wrapper:10:26)
    at Module._compile (internal/modules/cjs/loader.js:1118:30)
    at evalScript (internal/process/execution.js:94:25)
    at internal/main/eval_string.js:23:3
gyp: Call to 'node -e "console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/ ') + '/include/')"' returned exit status 1 while in binding.gyp. while trying to load binding.gyp
```

I am running foxy on ubuntu focal. On my native setup I can install and run it.  
This is the Dockerfile I am using. It may help others too.  

```
FROM ubuntu:focal
# distro download link has to be adopted also if you change the distro var
ENV ROS_DISTRO=foxy
ENV GIT_USER_NAME mrbuild
ENV GIT_USER_EMAIL mrbuild@github.com

RUN apt update && apt install -y git locales python curl wget
RUN locale-gen en_US en_US.UTF-8 && update-locale LC_ALL=en_US.UTF-8 LANG=en_US.UTF-8
ENV LANG en_US.UTF-8

RUN apt install -y gnupg2 lsb-release
RUN curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | apt-key add -
RUN sh -c 'echo "deb [arch=amd64,arm64] http://packages.ros.org/ros2/ubuntu `lsb_release -cs` main" > /etc/apt/sources.list.d/ros2-latest.list'

# Install prerequisites
RUN export DEBIAN_FRONTEND=noninteractive && apt update && apt install -y \
  build-essential \
  cmake \
  python3-colcon-common-extensions \
  python3-pip \
  python3-rosdep \
  libpython3-dev \
  cppcheck

RUN rosdep init
RUN rosdep update

# Configure git
RUN git config --global user.name $GIT_USER_NAME \
    && git config --global user.email $GIT_USER_EMAIL

# Get ROS2 latest package
ENV ROS2_WS=/root
WORKDIR $ROS2_WS

RUN wget https://github.com/ros2/ros2/releases/download/release-foxy-20200807/ros2-foxy-20200807-linux-focal-amd64.tar.bz2 \
    && tar xf ros2-foxy-20200807-linux-focal-amd64.tar.bz2

RUN rosdep install --from-paths $ROS2_WS/ros2-linux/share --ignore-src --rosdistro $ROS_DISTRO -y --skip-keys "console_bridge fastcdr fastrtps libopensplice67 libopensplice69 osrf_testing_tools_cpp poco_vendor rmw_connext_cpp rosidl_typesupport_connext_c rosidl_typesupport_connext_cpp rti-connext-dds-5.3.1 tinyxml_vendor tinyxml2_vendor urdfdom urdfdom_headers"

RUN echo "source $ROS2_WS/ros2-linux/local_setup.bash" >> $HOME/.bashrc

# Install Node.js
RUN curl -o- https://deb.nodesource.com/setup_12.x -o nodesource_setup.sh | bash
RUN apt install nodejs

RUN bash -c ". /opt/ros/$ROS_DISTRO/setup.bash"
RUN cd && git clone https://github.com/RobotWebTools/ros2-web-bridge.git && cd ros2-web-bridge && git checkout develop
WORKDIR /root/ros2-web-bridge/
# change rclnodejs version
RUN npm install -g dot-json && dot-json package.json dependencies.rclnodejs ^0.15.2
RUN npm install
CMD node bin/rosbridge.js

```

minggangw, 2020-08-10:
The error indicates that you haven't sourced the `local_setup.bash`, although you have `RUN bash -c ". /opt/ros/$ROS_DISTRO/setup.bash"
`. I suggest you could try something like:

``` bash
RUN echo "source $ROS2_WS/ros2-linux/local_setup.bash" >> $HOME/.bashrc
```

maxlein, 2020-08-11:
It's pretty much the Dockerfile from your repo.  
So the ROS2 binaries are downloaded and sourced.  

But I also tried from the foxy image itself -> `FROM ros:foxy`  
And also sourcing right before install like so:  
`RUN bash -c ". /opt/ros/$ROS_DISTRO/setup.sh" && npm install`  

And always getting the same error.

barnesew, 2020-08-18:
I am getting the same error. I tried sourcing ```$ROS2_WS/ros2-linux/local_setup.bash``` and ```/opt/ros/$ROS_DISTRO/setup.bash``` using both a Dockerfile and by overriding the entrypoint script.

minggangw, 2020-08-19:
Would you please add `echo $AMENT_PREFIX_PATH` into the Docker file before starting to install the `ros2-web-bridge` and see whether the variable has been set properly. So we could identify if the `rclnodejs` causes this error, thanks!

maxlein, 2020-09-03:
AMENT_PREFIX_PATH looks like that:  
```
echo $AMENT_PREFIX_PATH
/opt/ros/foxy

```

minggangw, 2020-09-03:
That's a little weird, from the error log:

```
console.log(process.env.AMENT_PREFIX_PATH.replace(/:/g, '/include/ ') + '/include/')
                                          ^

TypeError: Cannot read property 'replace' of undefined
```

It means `AMENT_PREFIX_PATH` is not a string,  which indicates the variable is not set yet. As the error only happens in Docker, I suspect it's caused by the configuration of docker?

minggangw, 2020-09-03:
The error originates from https://github.com/RobotWebTools/rclnodejs/blob/develop/binding.gyp#L63 when compiling with GYP file.

barnesew, 2020-09-30:
I was able to get it to work using this:
```
RUN /bin/bash -c "source \"/opt/ros/$ROS_DISTRO/setup.bash\" && npm install"
```

minggangw, 2020-10-02:
Great! @barnesew thanks for sharing your solution! As many developers met this issue when using Docker and reported the same issue on Github, would you please submit a PR to add a **Troubleshooting** section into [README](https://github.com/RobotWebTools/ros2-web-bridge/blob/develop/README.md) to describe the problem and how to solve it, I think that will help a lot, thanks!

closed 2021-01-01 by maxlein