RobotWebTools/ros2-web-bridge #154: Connection fail with browser

botamochi6277, 2020-09-26:
I have connection error to publish from browser.

I logged in  my ros pc (raspberry pi 4 installed foxy) with ssh and executed following command.

Terminal 1:
```
node bin/rosbridge.js
```

Terminal2:
```
cd examples && node index.js
```

When I opened `http://foxy.local:3000/html/publisher.html` from my host PC, the following page appeared. `foxy` is hostname of the raspberrypi.

[image]

Google Chrome said "WebSocket connection to 'ws://localhost:9090/' failed: Error in connection establishment: net::ERR_CONNECTION_REFUSED
Ros.connect @ roslib.js:2224
(anonymous) @ publisher.html:39".
[image]

minggangw, 2020-09-27:
Please try to change the address of ws server to your actual one, I think it should be the address of `raspberry pi 4`

https://github.com/RobotWebTools/ros2-web-bridge/blob/248f2ee44c525ebbd7c90e7c056ba62f3939eb02/examples/html/publisher.html#L39

This simple demo is designed to run the bridge and the web browser on the same machine.

botamochi6277, 2020-09-30:
Thank you for replying.
I succeed to publish when i run the bridge and the web browser on the same raspberry pi.
I fail to do, although, when i run the web browser on another pc with actual address (192.168.4.2).

minggangw, 2020-09-30:
> I fail to do, although, when i run the web browser on another pc with actual address (192.168.4.2).

What's the error log? still `net::ERR_CONNECTION_REFUSED`?

I suggest you could create a `WebSocket ` and try to connect to the address of raspberry pi, thus we could make sure it's reachable. Also, please run the bridge `DEBUG=ros2-web-bridge* node bin/rosbridge.js` to enable debug output and get more useful information.

botamochi6277, 2020-10-07:
I **succeed to publish** topics with the web browser from my Mac (external PC).
I edit line 40 in `examples/html/publisher.html` as below, 
`ros.connect('ws://localhost:9090');` -> `ros.connect('ws://foxy.local:9090');`.
`foxy` is hostname of my rospc running ros2-web-bridge.

Mac may try to connect 'ws://localhost:9090' as `'ws://[mac ip address]:9090'`.

minggangw, 2020-10-07:
Great! So could we close this issue now?

botamochi6277, 2020-10-07:
Yes, Please.

minggangw, 2020-10-07:
Close this issue, please feel free to reopen it if you still have problems, thanks!

closed 2020-10-07 by minggangw