RobotWebTools/ros2-web-bridge #155: Problem with converting web bridge into a binary with nexe

apoorvcn47, 2020-09-30:
When I try to convert rosbridge.js into a binary with nexe and put it in another directory, it cannot find binding and rclnodejs modules. While compiling, i do provide linking with node_modules, lib, package.json and index.js. Is there anything I am doing wrong? Following is the command I am using for compilation

minggangw, 2020-10-02:
Please `rclnodejs.node` is included.

apoorvcn47, 2020-10-02:
Unfortunately that did not work. Following is the error I am getting

/home/aw_rdex/Desktop/node_modules/bindings/bindings.js:211
      throw new Error(
      ^

Error: Could not find module root given file: "/home/aw_rdex/Desktop/node_modules/rclnodejs/lib/clock.js". Do you have a `package.json` file? 
    at Function.getRoot (/home/aw_rdex/Desktop/node_modules/bindings/bindings.js:211:13)
    at bindings (/home/aw_rdex/Desktop/node_modules/bindings/bindings.js:82:32)
    at Object.<anonymous> (/home/aw_rdex/Desktop/node_modules/rclnodejs/lib/clock.js:17:38)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)

I am including clock.js specifically. Following is my compilation command

nexe rosbridge2.js -r "./lib/*.js" -r "./node_modules/" -r "./index.js" -r "./package.json" -r "./node_modules/rclnodejs/build/Release/obj.target/rclnodejs.node" -r "./node_modules/bindings/*.js" -r "./node_modules/rclnodejs/lib/*.js" -r "./node_modules/rclnodejs/index.js" -r "./node_modules/rclnodejs/package.json" -r "./node_modules/rclnodejs/lib/clock.js"

minggangw, 2020-10-03:
The log shows the `rclnodejs.node` is not found, please check the source code of [clock.js](https://github.com/RobotWebTools/rclnodejs/blob/develop/lib/clock.js#L17)

```javascript
const rclnodejs = require('bindings')('rclnodejs');
```

apoorvcn47, 2020-10-05:
yes i understand that but based on my compilation command, it should be included, correct?

Sorry if I am being difficult, dont have much experience with javascript