RobotWebTools/ros2-web-bridge #157: serving Foxy + Gazebo11 to a Web Broswer

mis-eu, 2020-10-08:
Please validate if ros2-web-bridge supports deploying a ROS2 Foxy 20.04 Focal simulation via **gzclient**11 to a Web browser   (gzweb is still ROS1)

<!DOCTYPE html>
<html lang="en">
<head>
    <meta charset="UTF-8">
    <meta name="viewport" content="width=device-width, initial-scale=1.0">
    <title>Gazebo11 Web Bridge v 0.0.1</title>

    <link rel="stylesheet" href="css/base.css">

    <script type="text/javascript" src="http://static.robotwebtools.org/EventEmitter2/current/eventemitter2.min.js"></script>
    <script type="text/javascript" src="http://static.robotwebtools.org/roslibjs/current/roslib.min.js"></script>

</head>

<body>
   
    

    

    <script>
		let ros = new ROSLIB.Ros();
		let bridgeHost = window.location.hostname;
		let bridgePort = 9090;
    
    // monitor the connection    
    // listener for a connection event to the ros object. 
    // two blocks following do the same for error and close events. 
        
		ros.on('connection', function() {
    	console.log('bridge connected');
  	});
  	    ros.on('error', function(error) {
    	console.log('Error connecting to websocket server: ', error);
  	});
  	    ros.on('close', function() {
    	console.log('Connection to websocket server closed.');
  	});
        ros.connect('ws://' + bridgeHost + ':' + bridgePort);
       

    </script>  
    

    <iframe id="iframeGazebo" src="http://172.31.159.229:11345"></iframe>
  
    
</body>
</html>

minggangw, 2020-10-09:
I never try to combine them together, my first thought is that as **ROSLIB** doesn't aware if backed by ROS 1 or ROS 2, it seems feasible to interact gzweb with ros2-web-bridge by the ROSLIB.

mis-eu, 2020-10-12:
thank you for this response @minggangw, about the ROSLIB being agnostic to ros1/ros2. 

gzweb does not support compilation on Foxy/Focal as of yet.

To your knowledge, can ros2-web-bridge serve Foxy/Gazebo11 to a Browser? 
Would the server and client side code just connect to the Public IP node of Gazebo11, 
or do all the Physics and GUI messages and libraries have to be served as individual topic subscriptions?