RobotWebTools/ros2-web-bridge #158: Subscription Failure

edesta-be, 2020-10-28:
Hi everyone!
I was running the simple example and experimenting with different types of ROS messages when I had a failure trying to subscribe to a Joy data. 

I am attaching my publisher.html and subscription.html for your to recreate.

For your reference -- the error ONLY occurs on subscription. without trying to subscribe to the Joystick data the publisher is able to happily publish the data with no problem (and i can echo it out in my terminal).

 But this is my error below:
`#
# Fatal error in , line 0
# Check failed: length == backing_store->byte_length().
#
#
#
#FailureMessage Object: 0x7fff5bb2cc10
 1: 0xacbb41  [node]
 2: 0x1aa66d4 V8_Fatal(char const*, ...) [node]
 3: 0xf23bba v8::internal::GlobalBackingStoreRegistry::Lookup(void*, unsigned long) [node]
 4: 0xc6256f  [node]
 5: 0xc62774 v8::ArrayBuffer::New(v8::Isolate*, void*, unsigned long, v8::ArrayBufferCreationMode) [node]
 6: 0x7f2ae45014af rclnodejs::CreateArrayBufferFromAddress(Nan::FunctionCallbackInfo<v8::Value> const&) [/home/eden/Desktop/nvs/webapp_ws/ros2-web-bridge/node_modules/rclnodejs/build/Release/rclnodejs.node]
 7: 0x7f2ae44f29ac  [/home/eden/Desktop/nvs/webapp_ws/ros2-web-bridge/node_modules/rclnodejs/build/Release/rclnodejs.node]
 8: 0xca1cfb  [node]
 9: 0xca32ac  [node]
10: 0xca3926 v8::internal::Builtin_HandleApiCall(int, unsigned long*, v8::internal::Isolate*) [node]
11: 0x14c19b9  [node]
Illegal instruction (core dumped)
`

publisher.html file
`<!DOCTYPE html>
<html>
<head>
<meta charset="utf-8" />
<script src="https://static.robotwebtools.org/EventEmitter2/current/eventemitter2.min.js"></script>
<script src="https://static.robotwebtools.org/roslibjs/current/roslib.js"></script>

<script>
  // Connecting to ROS
  // -----------------
  const ros = new ROSLIB.Ros();

  const twist = {
    linear: {x: 0.2, y: 0.0, z: 0.5},
    angular: {x: 0.7, y: 1.5, z: 23.1},
  };
  
  const temp = {
    header: {
      stamp: {sec: 1603917727, nanosec: 208370251},
      frame_id: "temp",
    },
    temperature: 5.3,
    variance: 111.5,
  };

  const joy_data = {
    header: {
      stamp: {sec: 1603917727, nanosec: 208370251},
      frame_id: "joy",
    },
    axes: [0.2, 1.5, 8.9, 3.1],
    buttons: [1, 32],
  }

  // If there is an error on the backend, an 'error' emit will be emitted.
  ros.on('error', function(error) {
    document.getElementById('connecting').style.display = 'none';
    document.getElementById('connected').style.display = 'none';
    document.getElementById('closed').style.display = 'none';
    document.getElementById('error').style.display = 'inline';
    console.log(error);
  });

  // Find out exactly when we made a connection.
  ros.on('connection', function() {
    console.log('Connection made!');
    document.getElementById('connecting').style.display = 'none';
    document.getElementById('error').style.display = 'none';
    document.getElementById('closed').style.display = 'none';
    document.getElementById('connected').style.display = 'inline';
  });

  ros.on('close', function() {
    console.log('Connection closed.');
    document.getElementById('connecting').style.display = 'none';
    document.getElementById('connected').style.display = 'none';
    document.getElementById('closed').style.display = 'inline';
  });

  // Create a connection to the rosbridge WebSocket server.
  ros.connect('ws://localhost:9090');

  const temp_publisher = new ROSLIB.Topic({
    ros: ros,
    name : '/sense_temp',
    messageType : 'sensor_msgs/Temperature',
  })
  const publisher = new ROSLIB.Topic({
    ros: ros,
    name: '/example_twist',
    messageType : 'geometry_msgs/Twist',
  });
  const joy_publisher = new ROSLIB.Topic({
    ros: ros,
    name: '/joy_data',
    messageType : 'sensor_msgs/Joy',
  });

  setInterval(() => {
    publisher.publish(twist);
    temp_publisher.publish(temp);
    joy_publisher.publish(joy_data);
    document.getElementById("publisher").innerText = joy_data.header.frame_id;
  }, 1000);
</script>
</head>

<body>
  <h1>Simple Publisher Example</h1>
  <p>This example will pubilish a topic named "example_topic".</p>
  <div id="statusIndicator">
    <p id="connecting">
      Connecting to rosbridge...
    </p>
    <p id="connected" style="color:#00D600; display:none">
      Connected
    </p>
    <p id="error" style="color:#FF0000; display:none">
      Error in the backend!
    </p>
    <p id="closed" style="display:none">
      Connection closed.
    </p>
  </div>
  <div>
    <p>
      <b>Publish message: </b>
      <span id="publisher"></span>
    </p>
  </div>
</body>
</html>
`

**subscription.html file**:
`<!DOCTYPE html>
<html>
<head>
<meta charset="utf-8" />
<script src="https://static.robotwebtools.org/EventEmitter2/current/eventemitter2.min.js"></script>
<script src="https://static.robotwebtools.org/roslibjs/current/roslib.js"></script>

<script>
  // Connecting to ROS
  // -----------------
  var ros = new ROSLIB.Ros();

  // If there is an error on the backend, an 'error' emit will be emitted.
  ros.on('error', function(error) {
    document.getElementById('connecting').style.display = 'none';
    document.getElementById('connected').style.display = 'none';
    document.getElementById('closed').style.display = 'none';
    document.getElementById('error').style.display = 'inline';
    console.log(error);
  });

  // Find out exactly when we made a connection.
  ros.on('connection', function() {
    console.log('Connection made!');
    document.getElementById('connecting').style.display = 'none';
    document.getElementById('error').style.display = 'none';
    document.getElementById('closed').style.display = 'none';
    document.getElementById('connected').style.display = 'inline';
  });

  ros.on('close', function() {
    console.log('Connection closed.');
    document.getElementById('connecting').style.display = 'none';
    document.getElementById('connected').style.display = 'none';
    document.getElementById('closed').style.display = 'inline';
  });

  // Create a connection to the rosbridge WebSocket server.
  ros.connect('ws://localhost:9090');

  const example = new ROSLIB.Topic({
    ros: ros,
    name : '/sense_temp',
    messageType : 'sensor_msgs/Temperature',
  })

  // Subscribe a Topic
  example.subscribe(function(message) {
    document.getElementById("temperature").innerText = message.temperature
    document.getElementById("variance").innerText = message.variance
  });

  const joy_publisher = new ROSLIB.Topic({
    ros: ros,
    name: '/joy_data',
    messageType : 'sensor_msgs/Joy',
  });

  joy_publisher.subscribe(function(message) {
    document.getElementById("header").innerText = message.header.frame_id
  });

</script>
</head>

<body>
  <h1>Simple Subscription Example</h1>
  <p>This example will subscribe a topic named "example_topic".</p>
  <div id="statusIndicator">
    <p id="connecting">
      Connecting to rosbridge...
    </p>
    <p id="connected" style="color:#00D600; display:none">
      Connected
    </p>
    <p id="error" style="color:#FF0000; display:none">
      Error in the backend!
    </p>
    <p id="closed" style="display:none">
      Connection closed.
    </p>
  </div>
  <div>
    <b>Temperature: </b>
    <span id="temperature"></span>
  </div>
  <div>
    <b>Variance: </b>
    <span id="variance"></span>
  </div>
  <div>
    <b>Frame ID: </b>
    <span id="header"></span>
  </div>
</body>
</html>
`

minggangw, 2020-10-29:
Hi @edesta-be thanks for reporting the issue, would you please offer your detailed env, e.g. ROS2/nodejs/rclnodejs version? I try to reproduce it on my local Linux host, but it seems workable.

My env is:
 - Node.js: v12.18.4
 - ROS 2 Foxy Fitzroy - [Patch Release 3](https://github.com/ros2/ros2/releases/tag/release-foxy-20201009)
 - Ubuntu 20.04
 - rclnodejs [v0.16.0](https://www.npmjs.com/package/rclnodejs)

edesta-be, 2020-10-29:
Env:
Node.js: v15.0.1
ROS2 Foxy Fitzroy -- Patch Release 3
Ubuntu 20.04
rclnodejs v0.16.0

minggangw, 2020-10-29:
Because [rclnodejs](https://github.com/RobotWebTools/rclnodejs), ROS2 Node.js client, hasn't been verified on Node.js higher than 12, I suggest you could downgrade your Node.js. Actually, there is some known issue (https://github.com/RobotWebTools/rclnodejs/issues/713) on the higher version of Node.js

edesta-be, 2020-10-29:
Hi! Thank you for your help! I downgraded to 12.19.0 Node.js and it worked! 

It might be worth investigating still but for the time being I am all set! Thank you for your time!

closed 2020-10-29 by edesta-be

minggangw, 2020-10-30:
My pleasure, I will update here once it works on Node.js v14.x and v15.x