RobotWebTools/ros2-web-bridge #159: Segementation Fault

edesta-be, 2020-10-29:
Hi Folks!

I am getting a segmentation fault with the webbridge. It happens rather randomly I'm not quite sure how to go about tracking its problem. For your reference, all I am currently doing is subscribing to my joystick data that I am publishing via a Joy node (this is separate and running ROS2 foxy). 

```
Websocket started on ws://localhost:9090
(node:55671) DeprecationWarning: Deep requiring like `const uuidv4 = require('uuid/v4');` is deprecated as of uuid@7.x. Please require the top-level module when using the Node.js CommonJS module or use ECMAScript Modules when bundling for the browser. See https://github.com/uuidjs/uuid#deep-requires-now-deprecated for more information.
Segmentation fault (core dumped)
```

Is there something I am missing? Any help would be great! Thank you!

minggangw, 2020-10-30:
If you are using Node.js v12.19.0, I suspect there is an [issue](https://github.com/nodejs/node/issues/35620) which may cause the segfault and has been fixed. So please try v12.18.4 or wait for the next release (v12.19.1?). Also, this problem is tracked https://github.com/RobotWebTools/rclnodejs/issues/713 on rclnodejs.

edesta-be, 2020-10-30:
Thank you @minggangw that seems to solve the problem! I appreciate your response!

closed 2020-10-30 by edesta-be