RobotWebTools/ros2-web-bridge #160: Pose stamped message not published

DentOpt, 2020-12-07:
Hi y'all,

I am using your minimum example to build a publisher for a pose stamped message. 
I can publishing an empty message, a twist message, but no Pose and no Stamped message.  
It is creating their publishers as I can echo the topic, but nothing is written into the Pose and PoseStamped topics. 
Either my way of defining the topic message variable is wrong:
 ```
var posestamped_msg = new ROSLIB.Topic({
            header: {
              stamp: {
                secs : 0, 
                nsecs : 100
              },
              frame_id : "world"               //tf_prefix+"/"+map_frame;
            },
            pose: {
              position: {
                x : 0.0,
                y : 0.0,
                z : 0.0
              },
              orientation: {
                x : 0.0,
                y : 0.0,
                z : 0.0,
                w : 1.0
              }
            }
  });
```
or this would is a bug  in ROSLIB.
You can find my example.html file bellow. 
It would be great if you could give me hint how to debug this issue. 
Thanks in advance!

`<!DOCTYPE html>
<html>
<head>
<meta charset="utf-8" />
<script src="https://static.robotwebtools.org/EventEmitter2/current/eventemitter2.min.js"></script>
<script src="https://static.robotwebtools.org/roslibjs/current/roslib.js"></script>

<script>
  // Connecting to ROS
  // -----------------
  var ros = new ROSLIB.Ros();

  // If there is an error on the backend, an 'error' emit will be emitted.
  ros.on('error', function(error) {
    document.getElementById('connecting').style.display = 'none';
    document.getElementById('connected').style.display = 'none';
    document.getElementById('closed').style.display = 'none';
    document.getElementById('error').style.display = 'inline';
    console.log(error);
  });

  // Find out exactly when we made a connection.
  ros.on('connection', function() {
    console.log('Connection made!');
    document.getElementById('connecting').style.display = 'none';
    document.getElementById('error').style.display = 'none';
    document.getElementById('closed').style.display = 'none';
    document.getElementById('connected').style.display = 'inline';
  });

  ros.on('close', function() {
    console.log('Connection closed.');
    document.getElementById('connecting').style.display = 'none';
    document.getElementById('connected').style.display = 'none';
    document.getElementById('closed').style.display = 'inline';
  });

  // Create a connection to the rosbridge WebSocket server.
  ros.connect('ws://localhost:9090');

  // Publish a Topic
  var posestamped_publisher = new ROSLIB.Topic({
    ros : ros,
    name : '/posestamped',
    messageType : 'geometry_msgs/PoseStamped'
  });

  var posestamped_msg = new ROSLIB.Topic({
            header: {
              stamp: {
                secs : 0, 
                nsecs : 100
              },
              frame_id : "world"               //tf_prefix+"/"+map_frame;
            },
            pose: {
              position: {
                x : 0.0,
                y : 0.0,
                z : 0.0
              },
              orientation: {
                x : 0.0,
                y : 0.0,
                z : 0.0,
                w : 1.0
              }
            }
  });

  // Publish a Topic
  var pose_publisher = new ROSLIB.Topic({
    ros : ros,
    name : '/pose',
    messageType : 'geometry_msgs/Pose'
  });

  var pose_msg = new ROSLIB.Topic({
    position: {
      x : 0.0,
      y : 0.0,
      z : 0.0
    },
    orientation: {
      x : 0.0,
      y : 0.0,
      z : 0.0,
      w : 1.0
    }        
  });

  var twist_publisher = new ROSLIB.Topic({
    ros : ros,
    name : '/twist',
    messageType : 'geometry_msgs/Twist'
  });

  var twist_msg = new ROSLIB.Message({
    linear: {
      x: 0,
      y: 1,
      z: 0
    },
    angular: {
      x: 0,
      y: 1,
      z: 0
    }
  });
  
  
  var empty_publisher = new ROSLIB.Topic({
    ros : ros,
    name : '/empty',
    messageType : 'std_msgs/Empty'
  });

  var empty_msg = new ROSLIB.Message({});
  
  

  var count = 0;
  setInterval(() => {
    var msg = 'hello from ros2bridge ' + count++;
    
    posestamped_publisher.publish(posestamped_msg);
    pose_publisher.publish(pose_msg);
    twist_publisher.publish(twist_msg);
    empty_publisher.publish(empty_msg);
    document.getElementById("publisher").innerText = count;
  }, 1000);
</script>
</head>

<body>
  <h1>Simple Publisher Example</h1>
  <p>This example will pubilish a topic named "example_topic".</p>
  <div id="statusIndicator">
    <p id="connecting">
      Connecting to rosbridge...
    </p>
    <p id="connected" style="color:#00D600; display:none">
      Connected
    </p>
    <p id="error" style="color:#FF0000; display:none">
      Error in the backend!
    </p>
    <p id="closed" style="display:none">
      Connection closed.
    </p>
  </div>
  <div>
    <p>
      <b>Publish message: </b>
      <span id="publisher"></span>
    </p>
  </div>
</body>
</html>`

mis-eu, 2020-12-09:
I can confirm the same problem, the /topic is listed,
but the PoseStamped message echo does not appear

Could anyone please verify **PoseStamped** JSON structure below is valid? 
(slightly linited from above post, with added sequence int)

```
  let PoseStamped = new ROSLIB.Message({
            header: {
              seq: 1,
              stamp: {
                  secs: 0,
                  nsecs: 100
              },
              frame_id: "world"               
            },
            pose: {
                position: {
                    x: 0.0,
                    y: 0.0,
                    z: 0.0
                },
                orientation: {
                    x: 0.0,
                    y: 0.0,
                    z: 0.0,
                    w: 1.0
                }
            }
        });
```

mis-eu, 2020-12-09:
I can confirm the same problem, the /topic is listed,
but the PoseStamped message echo does not appear

Could anyone please verify **PoseStamped** JSON structure below is valid? 
(slightly linted from above post, with added sequence int)

```
  let PoseStamped = new ROSLIB.Message({
            header: {
              seq: 1,
              stamp: {
                  secs: 0,
                  nsecs: 100
              },
              frame_id: "world"               
            },
            pose: {
                position: {
                    x: 0.0,
                    y: 0.0,
                    z: 0.0
                },
                orientation: {
                    x: 0.0,
                    y: 0.0,
                    z: 0.0,
                    w: 1.0
                }
            }
        });
```

minggangw, 2020-12-09:
Sorry for being late, would you please try

```
DEBUG=ros2-web-bridge* node bin/rosbridge.js
```

to see any useful debug information?

mis-eu, 2020-12-09:
Thanks for your response @minggangw 

<details>
~/ros2-web-bridge$ DEBUG=ros2-web-bridge* node bin/rosbridge.js
  ros2-web-bridge:index ROS2 node started +0ms
  ros2-web-bridge:index Starting server on port 9090 +1ms
Websocket started on ws://localhost:9090
  ros2-web-bridge:Bridge Status level set to error (0) +0ms
  ros2-web-bridge:Bridge Web bridge de9db5da-ecd4-4149-98ae-d8f3b6af7173 is created +0ms
(node:24134) DeprecationWarning: Deep requiring like `const uuidv4 = require('uuid/v4');` is deprecated as of uuid@7.x. Please require the top-level module when using the Node.js CommonJS module or use ECMAScript Modules when bundling for the browser. See https://github.com/uuidjs/uuid#deep-requires-now-deprecated for more information.
  ros2-web-bridge:Bridge JSON command received: {"op":"advertise","id":"advertise:/move_base_simple/goal:1","type":"geometry_msgs/PoseStamped","topic":"/move_base_simple/goal","latch":false,"queue_size":100} +991ms
  ros2-web-bridge:Bridge advertise a topic: /move_base_simple/goal +1ms
  ros2-web-bridge:ResourceProvider Publisher has been created, and the topic name is /move_base_simple/goal. +0ms
  ros2-web-bridge:Bridge Suppressed: {"op":"status","level":"none","msg":"OK","id":"advertise:/move_base_simple/goal:1"} +8ms
  ros2-web-bridge:Bridge JSON command received: {"op":"publish","id":"publish:/move_base_simple/goal:2","topic":"/move_base_simple/goal","msg":{"isAdvertised":false,"compression":"none","throttle_rate":0,"latch":false,"queue_size":100,"queue_length":0,"reconnect_on_close":true},"latch":false} +35ms
  ros2-web-bridge:Bridge Publish a topic named /move_base_simple/goal with {"isAdvertised":false,"compression":"none","throttle_rate":0,"latch":false,"queue_size":100,"queue_length":0,"reconnect_on_close":true} +0ms
  ros2-web-bridge:Bridge Response: {"op":"status","level":"error","msg":"publish: TypeError: Invalid argument: frame_id in Header","id":"publish:/move_base_simple/goal:2"} +1ms
  ros2-web-bridge:Bridge JSON command received: {"op":"advertise","id":"advertise:/pose:3","type":"geometry_msgs/Pose","topic":"/pose","latch":false,"queue_size":100} +1ms
  ros2-web-bridge:Bridge advertise a topic: /pose +0ms
  ros2-web-bridge:ResourceProvider Publisher has been created, and the topic name is /pose. +39ms
  ros2-web-bridge:Bridge Suppressed: {"op":"status","level":"none","msg":"OK","id":"advertise:/pose:3"} +1ms
  ros2-web-bridge:Bridge JSON command received: {"op":"publish","id":"publish:/pose:4","topic":"/pose","msg":{"isAdvertised":false,"compression":"none","throttle_rate":0,"latch":false,"queue_size":100,"queue_length":0,"reconnect_on_close":true},"latch":false} +0ms
  ros2-web-bridge:Bridge Publish a topic named /pose with {"isAdvertised":false,"compression":"none","throttle_rate":0,"latch":false,"queue_size":100,"queue_length":0,"reconnect_on_close":true} +0ms
  ros2-web-bridge:Bridge Response: {"op":"status","level":"error","msg":"publish: TypeError: Invalid argument: x in Point","id":"publish:/pose:4"} +0ms
  ros2-web-bridge:Bridge JSON command received: {"op":"advertise","id":"advertise:/twist:5","type":"geometry_msgs/Twist","topic":"/twist","latch":false,"queue_size":100} +1ms
  ros2-web-bridge:Bridge advertise a topic: /twist +0ms
  ros2-web-bridge:ResourceProvider Publisher has been created, and the topic name is /twist. +3ms
  ros2-web-bridge:Bridge Suppressed: {"op":"status","level":"none","msg":"OK","id":"advertise:/twist:5"} +2ms
  ros2-web-bridge:Bridge JSON command received: {"op":"publish","id":"publish:/twist:6","topic":"/twist","msg":{"linear":{"x":0,"y":1,"z":0},"angular":{"x":0,"y":1,"z":0}},"latch":false} +0ms
  ros2-web-bridge:Bridge Publish a topic named /twist with {"linear":{"x":0,"y":1,"z":0},"angular":{"x":0,"y":1,"z":0}} +0ms
  ros2-web-bridge:Bridge Suppressed: {"op":"status","level":"none","msg":"OK","id":"publish:/twist:6"} +1ms
  ros2-web-bridge:Bridge JSON command received: {"op":"advertise","id":"advertise:/empty:7","type":"std_msgs/Empty","topic":"/empty","latch":false,"queue_size":100} +0ms

minggangw, 2020-12-09:
I noticed there was an error

```bash
ros2-web-bridge:Bridge JSON command received: {"op":"publish","id":"publish:/pose:4","topic":"/pose","msg":{"isAdvertised":false,"compression":"none","throttle_rate":0,"latch":false,"queue_size":100,"queue_length":0,"reconnect_on_close":true},"latch":false} +0ms ros2-web-bridge:Bridge Publish a topic named /pose with {"isAdvertised":false,"compression":"none","throttle_rate":0,"latch":false,"queue_size":100,"queue_length":0,"reconnect_on_close":true} +0ms ros2-web-bridge:Bridge Response: {"op":"status","level":"error","msg":"publish: TypeError: Invalid argument: x in Point","id":"publish:/pose:4"} +0ms
```

mis-eu, 2020-12-09:
error in PoseStamped is `Invalid argument: frame_id in Header","id":"publish:/move_base_simple/goal:2"}`

error in Pose is `Invalid argument: x in Point","id":"publish:/pose:4"}`

do the errors originate from JSON?

/PoseStamped
```
  var PoseStamped = new ROSLIB.Topic({
            header: {
              stamp: {
                secs : 0,
                nsecs : 100
              },
              frame_id : "world"               //tf_prefix+"/"+map_frame;
            },
            pose: {
              position: {
                x : 0.0,
                y : 0.0,
                z : 0.0
              },
              orientation: {
                x : 0.0,
                y : 0.0,
                z : 0.0,
                w : 1.0
              }
            }
  });
```

/pose
```
var pose_publisher = new ROSLIB.Topic({
    ros : ros,
    name : '/pose',
    messageType : 'geometry_msgs/Pose'
  });

  var pose_msg = new ROSLIB.Topic({
    position: {
      x: 0.0,
      y: 0.0,
      z: 0.0
    },
    orientation: {
      x: 0.0,
      y: 0.0,
      z: 0.0,
      w: 1.0
    }
  });
```

minggangw, 2020-12-09:
The `Time.msg` is ROS2 is different from that of ROS1, please see https://github.com/ros2/rcl_interfaces/blob/master/builtin_interfaces/msg/Time.msg, please reference the [example](https://github.com/RobotWebTools/rclnodejs/blob/develop/example/publisher-message-example.js#L32) to see if it will work, thanks!

mis-eu, 2020-12-09:
thanks for the information!

unfortunately, the Bridge response Error remains exactly the same as before. 
I´ve matched the [ROS2 Time.msg definition](https://github.com/RobotWebTools/rclnodejs/blob/45d7bd0a5cbc7b9d5bd6b397b33abc770845771b/example/publisher-message-example.js#L32) you referenced above,
as well as tried variations in frame id:  main frame, world frame, map frame. 

No dice. 

Since the [PoseStamped documentation ](http://docs.ros.org/en/noetic/api/geometry_msgs/html/msg/PoseStamped.html) generated in October 2020 is still for ROS1 Noetic, and deviates in structure from the ROS2 Time.msg
could it be a missing package migration in ROS2 Foxy?

mis-eu, 2020-12-09:
Sharing the complete code  for verification:

`<!DOCTYPE html>
<html>
<head>
<meta charset="utf-8" />
<script src="https://static.robotwebtools.org/EventEmitter2/current/eventemitter2.min.js"></script>
<script src="https://static.robotwebtools.org/roslibjs/current/roslib.js"></script>

<script>
  // Connecting to ROS
  // -----------------
  var ros = new ROSLIB.Ros();

  // If there is an error on the backend, an 'error' emit will be emitted.
  ros.on('error', function(error) {
    document.getElementById('connecting').style.display = 'none';
    document.getElementById('connected').style.display = 'none';
    document.getElementById('closed').style.display = 'none';
    document.getElementById('error').style.display = 'inline';
    console.log(error);
  });

  // Find out exactly when we made a connection.
  ros.on('connection', function() {
    console.log('Connection made!');
    document.getElementById('connecting').style.display = 'none';
    document.getElementById('error').style.display = 'none';
    document.getElementById('closed').style.display = 'none';
    document.getElementById('connected').style.display = 'inline';
  });

  ros.on('close', function() {
    console.log('Connection closed.');
    document.getElementById('connecting').style.display = 'none';
    document.getElementById('connected').style.display = 'none';
    document.getElementById('closed').style.display = 'inline';
  });

  // Create a connection to the rosbridge WebSocket server.
  ros.connect('ws://localhost:9090');

  // Publish a Topic
  var PoseStampedTopic = new ROSLIB.Topic({
    ros : ros,
    name : '/move_base_simple/goal',
    messageType : 'geometry_msgs/PoseStamped'
  });

/* ROS1 definition
  let PoseStamped = new ROSLIB.Message({
            header: {
              seq: 0,
              stamp: {
                  secs: 0,
                  nsecs: 100
              },
              frame_id: "world"               //tf_prefix+"/"+map_frame;
            },
            pose: {
                position: {
                    x: 0.0,
                    y: 0.0,
                    z: 0.0
                },
                orientation: {
                    x: 0.0,
                    y: 0.0,
                    z: 0.0,
                    w: 1.0
                }
            }
        });
*/

  var PoseStamped = new ROSLIB.Topic({
            header: {
              stamp: {
                sec: 0,
                nanosec: 100
              },
              frame_id: 'main frame'
            },
            pose: {
              position: {
                x: 0.0,
                y: 0.0,
                z: 0.0
              },
              orientation: {
                x: 0.0,
                y: 0.0,
                z: 0.0,
                w: 1.0
              }
            }
  });

  /*ROS2 time message
        header: {
          stamp: {
            sec: 123456,
            nanosec: 789,
          },
          frame_id: 'main frame',
        },
        name: ['Tom', 'Jerry'],
        position: [1, 2],
        velocity: [2, 3],
        effort: [4, 5, 6],
      });
      console.log(`Publish ${++count} messages.`);
    }, 1000);

    rclnodejs.spin(node);
  */

  /*DEBUG LOG
  ros2-web-bridge:Bridge Publish a topic named /move_base_simple/goal with {"isAdvertised":false,"compression":"none","throttle_rate":0,"latch":false,"queue_size":100,"queue_length":0,"reconnect_on_close":true} +1ms

  ros2-web-bridge:Bridge Response: {"op":"status","level":"error","msg":"publish: TypeError: Invalid argument: frame_id in Header","id":"publish:/move_base_simple/goal:77"} +0ms
  */

  // Publish a Topic
  var PoseTopic = new ROSLIB.Topic({
    ros : ros,
    name : '/pose',
    messageType : 'geometry_msgs/Pose'
  });

  var pose_msg = new ROSLIB.Topic({
    position: {
      x: 0.0,
      y: 0.0,
      z: 0.0
    },
    orientation: {
      x: 0.0,
      y: 0.0,
      z: 0.0,
      w: 1.0
    }
  });

  var twist_publisher = new ROSLIB.Topic({
    ros : ros,
    name : '/twist',
    messageType : 'geometry_msgs/Twist'
  });

  var twist_msg = new ROSLIB.Message({
    linear: {
      x: 0,
      y: 1,
      z: 0
    },
    angular: {
      x: 0,
      y: 1,
      z: 0
    }
  });

  var empty_publisher = new ROSLIB.Topic({
    ros : ros,
    name : '/empty',
    messageType : 'std_msgs/Empty'
  });

  var empty_msg = new ROSLIB.Message({});

  var count = 0;
  setInterval(() => {
    var msg = 'hello from ros2bridge ' + count++;

    PoseStampedTopic.publish(PoseStamped);
    PoseTopic.publish(pose_msg);
    twist_publisher.publish(twist_msg);
    empty_publisher.publish(empty_msg);
    document.getElementById("publisher").innerText = count;
  }, 1000);
</script>
</head>

<body>
  <h1>Simple Publisher Example</h1>
  <p>This example will pubilish a topic named "example_topic".</p>
  <div id="statusIndicator">
    <p id="connecting">
      Connecting to rosbridge...
    </p>
    <p id="connected" style="color:#00D600; display:none">
      Connected
    </p>
    <p id="error" style="color:#FF0000; display:none">
      Error in the backend!
    </p>
    <p id="closed" style="display:none">
      Connection closed.
    </p>
  </div>
  <div>
    <p>
      <b>Publish message: </b>
      <span id="publisher"></span>
    </p>
  </div>
</body>
</html>`

minggangw, 2020-12-09:
```json
  ros2-web-bridge:Bridge Publish a topic named /move_base_simple/goal with {"isAdvertised":false,"compression":"none","throttle_rate":0,"latch":false,"queue_size":100,"queue_length":0,"reconnect_on_close":true} +1ms
  ros2-web-bridge:Bridge Response: {"op":"status","level":"error","msg":"publish: TypeError: Invalid argument: frame_id in Header","id":"publish:/move_base_simple/goal:77"} +0ms
```
The JSON string that the bridge received is 
```json
{"isAdvertised":false,"compression":"none","throttle_rate":0,"latch":false,"queue_size":100,"queue_length":0,"reconnect_on_close":true} 
```
I think that is why there was an error thrown, this JSON is supposed to carry `PoseStamped` object, apparently, it was not. (The string was sent by ROSLib through websockets)

mis-eu, 2020-12-09:
@minggangw in the absence of documentation, how would you test publishing **PoseStamped** via JSON ?

mis-eu, 2020-12-09:
@minggangw 
otherwise, how can the rclnodejs publisher be implemented in clientside HTML, where _`require`_ is not defined? I tried importing rclnodejs as a module, but it was not possible to export it as such within ros2-web-bridge/node_moeules/rclnodejs/index.js   

```
        const rclnodejs = require('rclnodejs');
        rclnodejs.init().then(() => {
          const node = rclnodejs.createNode('publisher_example_node');
          const publisher = node.createPublisher('std_msgs/msg/String', 'topic');
          publisher.publish(`Hello ROS 2 from rclnodejs`);
          rclnodejs.spin(node);
        });
```

mis-eu, 2020-12-10:
> The `Time.msg` is ROS2 is different from that of ROS1, please see https://github.com/ros2/rcl_interfaces/blob/master/builtin_interfaces/msg/Time.msg, please reference the [...]

Here is the above [Example](https://github.com/RobotWebTools/rclnodejs/blob/develop/example/publisher-message-example.js#L32) plugged into the [/demo/ index.html](https://github.com/RobotWebTools/ros2-web-bridge/tree/develop/demo)

The problem is that even when loaded into the HTML page, the same error is thrown, making  it not possible to spin the rclnode publisher: 

**publisher-message-example.js:19 Uncaught ReferenceError: require is not defined
    at publisher-message-example.js:19**

```
<!DOCTYPE html>
<html lang="en" class="no-js">
<head>
	<meta charset="UTF-8" />
	<meta http-equiv="X-UA-Compatible" content="IE=edge">
	<meta name="viewport" content="width=device-width, initial-scale=1">
	<title>ROS</title>
	<meta name="keywords" content="roslibjs, ros2-web-bridge and rclnodejs"	/>
	<meta name="author" content="ros" />
	<link rel="stylesheet" type="text/css" href="css/demo.css" />
	<link rel="stylesheet" type="text/css" href="css/component.css" />
	<script src="https://static.robotwebtools.org/EventEmitter2/current/eventemitter2.min.js"></script>
	<script src="https://static.robotwebtools.org/roslibjs/current/roslib.min.js"></script>
	<script src="https://static.robotwebtools.org/EaselJS/current/easeljs.min.js"></script>
	<script src="https://static.robotwebtools.org/ros2djs/current/ros2d.min.js"></script>
</head>
<body>
	<div id="perspective" class="perspective effect-movedown">
		<div class="container">
			<div class="wrapper">
				<header class="ros-header">
					<div class="columnleft">
						<div class='controller'>
							<div class='nx'>

					<div class="twodmap" id="ros2dmap">
                    </div>

                </header>

				<div class="main clearfix">
					<div class="log" id="log"></div>
                </div>

			</div>
		</div>
    </div>
    <!--
	<script src="js/btnstate.js"></script>
	<script src="js/ros2dmap.js"></script>
    <script src="js/controller.js"></script>
    -->
    <script src="publisher-message-example.js"></script>
	<script>
		let ros = new ROSLIB.Ros();
		let bridgeHost = window.location.hostname;
		let bridgePort = 9090;

		ros.on('connection', function() {
    	console.log('bridge connected');
  	});
  	ros.on('error', function(data) {
    	console.log('error');
  	});
  	ros.on('closed', function() {
    	console.log('bridge closed');
  	});
		ros.connect('ws://' + bridgeHost + ':' + bridgePort);

		let publisher = new rclnodejs(ros, {
			divName: 'log',
    	//width: 400,
    	//height: 400
		});
		//let webRosController = new WebRosController(ros, ros2dmap);
		//let btnState = new ButtonState();

	</script>
	</body>
</html>

```

minggangw, 2020-12-10:
Hi @mis-eu thanks for your continuous investigation on this issue. `require` is kind of CommonJS module concept, which the browser doesn't support directly. Some library, e.g. [browserify](https://github.com/browserify/browserify), can support it in your browser. But `rclnodejs` will load a C++ addon that has to use a Node.js, so you cannot import the module in your browser. My suggestion is:

-  If you have to use the browser to provide a GUI, I suggest you could leverage [electron](https://www.electronjs.org/) that can support Node.js in your web app, because as you know, [rosbridge v2.0 Protocol](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md) implemented by this `ros2-web-bridge` is designed for ROS 1, aslo the ROSLib library. So it's hard to support full features of ROS2, while the `rclnodejs` can support ROS2 well as a Node.js client.

- If you can use Node.js directly, I encourage you to migrate to it! [`rclnodejs`](https://github.com/RobotWebTools/rclnodejs) always tracks the latest stable version of ROS2. Hope it's helpful to you, thanks!

mis-eu, 2020-12-10:
Hello @minggangw 
thanks very much for the extensive [examples provided in rclnodejs](https://github.com/RobotWebTools/rclnodejs/tree/develop/example) package as well as the good [rclnodejs typescript use-cases documentation](https://github.com/RobotWebTools/rclnodejs#using-typescript). 

I suspected the C++ codebase could not be made available to the Browser without a [JavaScript loader](https://docs.google.com/spreadsheets/d/1Cc55nwxNQsdE9G7A4_GKhLgfSgUNyGibQY3AjuLAYKE/edit#gid=2). 

Thanks for confirming this and the [electron](https://www.electronjs.org/) recommendation, 
though the question remains why ROSlib is not tunneling the Pose and PoseStamped JSON object correctly.

minggangw, 2020-12-10:
Maybe you could log in the roslibjs at https://github.com/RobotWebTools/roslibjs/blob/969f568ceeb15600ebea1a5047bca5b34b7e58ad/src/util/workerSocketImpl.js#L27 to see if the JSON string is correct practically when it's being sent, and if you want to do more in ROS2 by JavaScript, using `rclnodejs` is a good idea in the long term :smile:

mis-eu, 2020-12-11:
It is interesting that Node is throwing the [same error rosbridge returned for the ROSlib JSON of PoseStamped](https://github.com/RobotWebTools/ros2-web-bridge/issues/160#issuecomment-741495961) discussed above.

Running node on the  [publisher-example.js](https://github.com/RobotWebTools/rclnodejs/blob/develop/example/publisher-example.js) also returns **TypeError: Invalid argument: frame_id in Header**

script:
$ node publisher-example-posestamped.js

output:
```
Publishing message: 0
/home/user/ros2-web-bridge/node_modules/rclnodejs/generated/std_msgs/std_msgs__msg__Header.js:71
        throw new TypeError('Invalid argument: frame_id in Header');
        ^

TypeError: Invalid argument: frame_id in Header
    at HeaderWrapper.freeze (/home/user/ros2-web-bridge/node_modules/rclnodejs/generated/std_msgs/std_msgs__msg__Header.js:71:15)
    at PoseStampedWrapper.freeze (/home/user/ros2-web-bridge/node_modules/rclnodejs/generated/geometry_msgs/geometry_msgs__msg__PoseStamped.js:69:32)
    at PoseStampedWrapper.serialize (/home/user/ros2-web-bridge/node_modules/rclnodejs/generated/geometry_msgs/geometry_msgs__msg__PoseStamped.js:75:10)
    at Publisher.publish (/home/user/ros2-web-bridge/node_modules/rclnodejs/lib/publisher.js:58:38)
    at Timeout.setInterval [as _onTimeout] (/home/user/ros2-web-bridge/publishnode_posestamped.js:26:15)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
```

mis-eu, 2020-12-11:
and running node on that same [publisher-example.js](https://github.com/RobotWebTools/rclnodejs/blob/develop/example/publisher-example.js) but with geometry_msgs/msg/Twist on /cmd_vel topic throws the same **Invalid argument: x in Vector 3 error**  as with roslibjs

output error:
```
(node:6128) UnhandledPromiseRejectionWarning: TypeError: Invalid argument: x in Vector3
    at Vector3Wrapper.freeze (/home/user/ros2-web-bridge/node_modules/rclnodejs/generated/geometry_msgs/geometry_msgs__msg__Vector3.js:70:15)
    at TwistWrapper.freeze (/home/user/ros2-web-bridge/node_modules/rclnodejs/generated/geometry_msgs/geometry_msgs__msg__Twist.js:68:32)
    at TwistWrapper.serialize (/home/user/ros2-web-bridge/node_modules/rclnodejs/generated/geometry_msgs/geometry_msgs__msg__Twist.js:74:10)
    at Publisher.publish (/home/user/ros2-web-bridge/node_modules/rclnodejs/lib/publisher.js:58:38)
    at rclnodejs.init.then (/home/user/ros2-web-bridge/publishnode_nobindings.js:19:13)
(node:6128) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:6128) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

script:
```
#!/usr/bin/env node

'use strict';

/* eslint-disable camelcase */

const rclnodejs = require('rclnodejs');
const debug = require('debug')('rclnodejs:publisher');
const Entity = require('./entity.js');

rclnodejs.init().then(() => {
  const node = rclnodejs.createNode('publisher_example_node');
  const publisher = node.createPublisher('geometry_msgs/msg/Twist', 'cmd_vel');
  publisher.publish('cmd_vel');
    console.log('cmd_vel');
  rclnodejs.spin(node);
});
```

minggangw, 2020-12-11:
I think the case is a little mess here,  let me clarify it. I tried the example below and it works, and you can verify by writing a subscriber to log the topic you received. My env is: Nodejs v12.20.0, rclnodejs v0.17.0, ROS2 Foxy patch release3.

```js
'use strict';

/* eslint-disable camelcase */

const rclnodejs = require('rclnodejs');

rclnodejs
  .init()
  .then(() => {
    const node = rclnodejs.createNode('publisher_message_example_node');
    const publisher = node.createPublisher(
      'geometry_msgs/msg/PoseStamped',
      'PoseStamped'
    );
    let count = 0;

    setInterval(function () {
      publisher.publish({
        header: {
          stamp: {
            sec : 0, 
            nanosec : 100
          },
          frame_id : "world"               //tf_prefix+"/"+map_frame;
        },
        pose: {
          position: {
            x : 0.0,
            y : 0.0,
            z : 0.0
          },
          orientation: {
            x : 0.0,
            y : 0.0,
            z : 0.0,
            w : 1.0
          }
        }
});
      console.log(`Publish ${++count} messages.`);
    }, 1000);

    rclnodejs.spin(node);
  })
  .catch((e) => {
    console.log(e);
  });
```

and to subscribe the topic
```js
'use strict';

const rclnodejs = require('../index.js');

rclnodejs
  .init()
  .then(() => {
    const node = rclnodejs.createNode('subscription_message_example_node');
    let count = 0;

    node.createSubscription(
      'geometry_msgs/msg/PoseStamped',
      'PoseStamped',
      (pose) => {
        console.log(`Received message No. ${++count}: `, pose);
      }
    );

    rclnodejs.spin(node);
  })
  .catch((e) => {
    console.log(e);
  });
```

mis-eu, 2020-12-13:
@minggangw Thank you!

These validate publishing of `PoseStamped` also on ROS2 env:  **Foxy (Debian)**, Node v10.19.0, rclnodejs v0.17.0

I had to make just one change in the _Subscriber script_, to require (rclnodejs) instead of (../index.js) as below: 

```
'use strict';

const rclnodejs = require('rclnodjejs');

rclnodejs
  .init()
  .then(() => {
    const node = rclnodejs.createNode('subscription_message_example_node');
    let count = 0;

    node.createSubscription(
      'geometry_msgs/msg/PoseStamped',
      'PoseStamped',
      (pose) => {
        console.log(`Received message No. ${++count}: `, pose);
      }
    );

    rclnodejs.spin(node);
  })
  .catch((e) => {
    console.log(e);
  });

```
[image]
[image]

Thank you again, have a nice week!

minggangw, 2020-12-14:
That's it 🚀