RobotWebTools/ros2-web-bridge #161: Start of rosbridge.js aborts (librcl.so not found)

whatis777, 2021-01-02:
When trying to start ros2-web-bridge with `node bin/rosbridge.js`, it aborts with this error message:
```
/home/ubuntu/webbridgetest/ros2-web-bridge/node_modules/bindings/bindings.js:121
        throw e;
        ^
Error: librcl.so: cannot open shared object file: No such file or directory
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1057:18)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Module.require (internal/modules/cjs/loader.js:887:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at bindings (/home/ubuntu/webbridgetest/ros2-web-bridge/node_modules/bindings/bindings.js:112:48)
    at Object.<anonymous> (/home/ubuntu/webbridgetest/ros2-web-bridge/node_modules/rclnodejs/lib/clock.js:17:38)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
```

The setup is: 

- Ubuntu Server 20.04.1 LTS
- python 3.8.5
- ROS2 Foxy Fitzroy following [these instructions](https://roboticsbackend.com/install-ros2-on-raspberry-pi/)
- ros2-web-bridge 0.3.0
- nodejs 12.20.0
- npm 6.14.8
- hardware: raspberry pi 3 B v1.2

`librcl.so` is located in `/opt/ros/foxy/lib`:
`-rw-r--r--  1 root root  223912 Dec 10 11:48 librcl.so`

`echo $LD_LIBRARY_PATH` prints 
`/opt/ros/foxy/opt/yaml_cpp_vendor/lib:/opt/ros/foxy/lib/aarch64-linux-gnu:/opt/ros/foxy/lib`

The terminals are sourced by default (`echo "source /opt/ros/foxy/setup.bash" >> ~/.bashrc`)

ros2-web-bridge was installed in a directory besides the ROS2 workspace using these commands to install it:
```
git clone https://github.com/RobotWebTools/ros2-web-bridge.git
cd ros2-web-bridge
git checkout 0.3.0
npm install
```
Can you please help me here?

minggangw, 2021-01-02:
It seems that everything has been set properly, normally this kind of issue was caused by not sourcing the `setup.bash` for ROS2. Can you run the examples of rclpy?

whatis777, 2021-01-02:
I have a python-based custom node up & running properly  & publishing ROS2 messages if that helps.
If not please let me know which example I should run

minggangw, 2021-01-02:
I just want to confirm whether the `librcl.so` could be found because rclpy is similar to rclnodejs by wrapping the `librcl.so`  library. Currently, I have no idea what causes it, but I suggest you could try to use https://github.com/ros2/ros2/releases/download/release-foxy-20201211/ros2-foxy-20201211-linux-focal-arm64.tar.bz2 package to see if it works.

whatis777, 2021-01-02:
1. I did a ROS2 reinstall using the binary and this [tutorial]()
2. reboot 
3. reinstall ros2-web-bridge (performing the steps as described above)

Performing `npm install` already shows a similar error at the end of the build:
```
....
....
make: Leaving directory '/home/ubuntu/webbridgetest/ros2-web-bridge/node_modules/rclnodejs/build'
> rclnodejs@0.17.0 postinstall /home/ubuntu/webbridgetest/ros2-web-bridge/node_modules/rclnodejs
> node scripts/generate_messages.js
Start JavaScript message generation...
Caught error: Error: librcl.so: cannot open shared object file: No such file or directory
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@~2.1.1 (node_modules/chokidar/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm64"})
added 414 packages from 318 contributors and audited 415 packages in 396.636s
```