RobotWebTools/ros2-web-bridge #162: Error in generated service request files

Jopand, 2021-01-04:
I'm running a Debian 10 environment with ROS2 Foxy.
I tried to make a simple Angular page calling a change_state service from ROS. It didn't work.

After some digging and debugging, I found out that it was due to an error in ./lifecycle_msgs/lifecycle_msgs__srv__ChangeState_Request.js

[image]

The generated line 31 is:
**translator.constructFromPlanObject(this, other);**
If I send other.request as the 2nd argument, it all works:
translator.constructFromPlanObject(this, other.request);

This is not just the case with the ChangeState service, but also some other homemade services. The above fix works with these services as well

minggangw, 2021-01-05:
Please reference the example, I think it should work :)

https://github.com/RobotWebTools/ros2-web-bridge/blob/248f2ee44c525ebbd7c90e7c056ba62f3939eb02/examples/html/client.html#L42-L57

Jopand, 2021-01-05:
Problem seems to be in rclnodejs. So I created the issue there instead

Jopand, 2021-01-05:
Okay, I tried to use your example. I get no response at all.

To make it work, I have to call it like this

`var addTwoInts = new ROSLIB.Service({
  ros : $scope.ros,
  name : '/add_two_ints',
  serviceType : 'example_interfaces/AddTwoInts'
});

var request = new ROSLIB.ServiceRequest({
  a : 1,
  b : 2
});
var args = {request :request, type: 'example_interfaces/AddTwoInts'};
// Send the request
addTwoInts.callService(args, function(result) {
  console.log('Result for service call on ' + addTwoInts.name + ': ' + result.sum);
});`

Cannot make it work without the 'args'. Then I reach some breakpoints on the nodejs side, but then I also have to make the change in the generated code to get a response :-S What am I doing wrong?

minggangw, 2021-01-06:
Hi @Jopand I tested the example locally and it seem working well, so would you please run with 
```bash
DEBUG=ros2-web-bridge* node bin/rosbridge.js
```
to check the debug output? Here is mine

```bash
  ros2-web-bridge:index ROS2 node started +0ms
  ros2-web-bridge:index Starting server on port 9090 +1ms
Websocket started on ws://localhost:9090
  ros2-web-bridge:Bridge Status level set to error (0) +0ms
  ros2-web-bridge:Bridge Web bridge 2574badb-496f-4104-b366-d545dbc1ac89 is created +1ms
(node:1837130) DeprecationWarning: Deep requiring like `const uuidv4 = require('uuid/v4');` is deprecated as of uuid@7.x. Please require the top-level module when using the Node.js CommonJS module or use ECMAScript Modules when bundling for the browser. See https://github.com/uuidjs/uuid#deep-requires-now-deprecated for more information.
  ros2-web-bridge:Bridge JSON command received: {"op":"advertise_service","type":"example_interfaces/AddTwoInts","service":"/add_two_ints"} +11ms
  ros2-web-bridge:ResourceProvider Service has been created, and the service name is /add_two_ints. +0ms
  ros2-web-bridge:Bridge Suppressed: {"op":"status","level":"none","msg":"OK"} +11ms
  ros2-web-bridge:Bridge Status level set to error (0) +2s
  ros2-web-bridge:Bridge Web bridge 6c1a357f-45c2-44d5-b84e-4929bd7055a2 is created +0ms
  ros2-web-bridge:Bridge JSON command received: {"op":"call_service","id":"call_service:/add_two_ints:1","service":"/add_two_ints","type":"example_interfaces/AddTwoInts","args":{"a":1,"b":2}} +16ms
  ros2-web-bridge:ResourceProvider Client has been created, and the service name is /add_two_ints. +2s
  ros2-web-bridge:Bridge Suppressed: {"op":"status","level":"none","msg":"OK","id":"call_service:/add_two_ints:1"} +13ms
  ros2-web-bridge:Bridge JSON command received: {"op":"service_response","service":"/add_two_ints","values":{"sum":3},"result":true,"id":"e427f873-5ec2-485f-9090-3cdc5cd47295"} +3ms
  ros2-web-bridge:Bridge Suppressed: {"op":"status","level":"none","msg":"OK","id":"e427f873-5ec2-485f-9090-3cdc5cd47295"} +2ms

```

Jopand, 2021-01-08:
I just created a clean Ubuntu vm with ros2 foxy and couldn't reproduce the problem. So guess I have had a "dirty" setup before. Think we can close the issue

closed 2021-01-08 by Jopand