RobotWebTools/ros2-web-bridge #168: Using ros2-web-bridge for Node-Red implementation

a-pastore, 2021-01-11:
Hello all,

i would like to ask you if anyone here already has experience with the connection between ROS2 and node-red. I was hoping to use the ros2-web-bridge in combination with the following node-red-module to create a web-based joystick, for instance. 

[node-red-contrib-ros](https://flows.nodered.org/node/node-red-contrib-ros)

The following system is used in a VM:

- Ubuntu 20.04 LTS
- ROS2: Foxy Fitzroy
- ros2-web-bridge: 0.3.0
- node: 10.19.0
- npm: 6.14.11
- node-red: 1.2.2

Can someone please assist with the following confusion? Is it enough to directly access the websocket ws://127.0.0.1:9090 to access the ROS2 Topics? Or does the ros2-web-bridge differ in this respect from the ROSbridge based on [this tutorial](https://marvinferber.net/?p=14)? 

Thank you for your time

minggangw, 2021-01-12:
Theoretically, I think it's feasible to use `ros2-web-bridge` to replace the `ROS bridge` because  `ros2-web-bridge` is designed to be compatible with [roslibjs](https://github.com/RobotWebTools/roslibjs) and follows the [rosbridge v2 protocol](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md).

The warning you mentioned is not critical, please try to run the `ros2-web-bridge`:

```
DEBUG=ros2-web-bridge* node bin/rosbridge.js
```

to find any meaningful output. For Node.js, I suggest you upgrade to v12.20.1.

vkuehn, 2021-01-12:
@a-pastore interesting project !
It looks like you are using an older node version. Look here [Node Red Supported Versions](https://nodered.org/docs/faq/node-versions). 
Maybe you want to forward ROS2 as an issue to https://github.com/namgk/node-red-contrib-ros ?