RobotWebTools/ros2-web-bridge #171: Handling of /tf and /tf_static

AndrewJSchoen, 2021-01-24:
Hi all,

I was looking at using this library for visualizing some robots with a ros2 backend. I have gotten things to the point on the backend ros2 sever that the robot displays correctly in rviz2, but the robot does not display with the ros3djs visualizer. I looked into the connections that the `ros2_web_bridge` node has, and found that it does not subscribe to the `/tf` or `/tf_static`, despite setting up a TFClient (roslibjs). Is this something that I am doing incorrectly here, or is this not an intended behavior to have the subscriptions created?

Let me know if you need any relevant code and I will be happy to provide it.

Thanks!

AndrewJSchoen, 2021-01-27:
Just to connect some threads, I ended up creating a ROS2-compatible version of `tf2_web_republisher` but am still having issues with both `ros2-web-bridge` and `rosbridge_server`, documented in [rosbridge_suite#552](https://github.com/RobotWebTools/rosbridge_suite/issues/552)

minggangw, 2021-01-28:
Sorry for the delay, I suggest you could use

```
DEBUG=ros2-web-bridge* node bin/rosbridge.js
```

to check the log.

AndrewJSchoen, 2021-01-28:
Hi, thanks for the response.

I did that and got the following:

```
DEBUG=ros2-web-bridge* node bin/rosbridge.js
  ros2-web-bridge:index ROS2 node started +0ms
  ros2-web-bridge:index Starting server on port 9090 +1ms
Websocket started on ws://localhost:9090
  ros2-web-bridge:Bridge Status level set to error (0) +0ms
  ros2-web-bridge:Bridge Web bridge 055ddf70-f2de-4107-9f6f-c09b614065cc is created +0ms
(node:93228) DeprecationWarning: Deep requiring like `const uuidv4 = require('uuid/v4');` is deprecated as of uuid@7.x. Please require the top-level module when using the Node.js CommonJS module or use ECMAScript Modules when bundling for the browser. See https://github.com/uuidjs/uuid#deep-requires-now-deprecated for more information.
(Use `node --trace-deprecation ...` to show where the warning was created)
  ros2-web-bridge:Bridge JSON command received: {"op":"advertise","id":"advertise:/tf2_web_republisher/goal:15","type":"tf2_web_republisher/TFSubscriptionActionGoal","topic":"/tf2_web_republisher/goal","latch":false,"queue_size":100} +49ms
  ros2-web-bridge:Bridge advertise a topic: /tf2_web_republisher/goal +0ms
  ros2-web-bridge:Bridge Response: {"op":"status","level":"error","msg":"advertise: Error: The message required does not exist: tf2_web_republisher, msg, TFSubscriptionActionGoal","id":"advertise:/tf2_web_republisher/goal:15"} +0ms
  ros2-web-bridge:Bridge JSON command received: {"op":"advertise","id":"advertise:/tf2_web_republisher/cancel:16","type":"actionlib_msgs/GoalID","topic":"/tf2_web_republisher/cancel","latch":false,"queue_size":100} +2ms
  ros2-web-bridge:Bridge advertise a topic: /tf2_web_republisher/cancel +0ms
  ros2-web-bridge:ResourceProvider Publisher has been created, and the topic name is /tf2_web_republisher/cancel. +0ms
  ros2-web-bridge:Bridge Suppressed: {"op":"status","level":"none","msg":"OK","id":"advertise:/tf2_web_republisher/cancel:16"} +64ms
  ros2-web-bridge:Bridge JSON command received: {"op":"subscribe","id":"subscribe:/tf2_web_republisher/feedback:17","type":"tf2_web_republisher/TFSubscriptionActionFeedback","topic":"/tf2_web_republisher/feedback","compression":"none","throttle_rate":0,"queue_length":0} +0ms
  ros2-web-bridge:Bridge subscribe a topic named /tf2_web_republisher/feedback +0ms
  ros2-web-bridge:Bridge Response: {"op":"status","level":"error","msg":"subscribe: Error: The message required does not exist: tf2_web_republisher, msg, TFSubscriptionActionFeedback","id":"subscribe:/tf2_web_republisher/feedback:17"} +0ms
  ros2-web-bridge:Bridge JSON command received: {"op":"subscribe","id":"subscribe:/visualization_marker:18","type":"visualization_msgs/Marker","topic":"/visualization_marker","compression":"png","throttle_rate":0,"queue_length":0} +1ms
  ros2-web-bridge:Bridge subscribe a topic named /visualization_marker +0ms
```

Now, if I am reading this correctly, it seems like the issue is actually that roslibjs is looking for the wrong stuff in the wrong places (Expecting ROS1-formatted messages/services/actions instead of ROS2), and isn't actually an issue with ros2-web-bridge itself _per se_, unless it is trying to account for whatever protocol it expects. Thoughts?

minggangw, 2021-01-28:
The roslibjs doesn't support ROS2 officially, ros2-web-bridge depends on [rosbridge v2 protocol](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md) to communicate with the front-end JS library, which I think causes this issue.

ohsirawat, 2021-02-07:
This `"msg":"advertise: Error: The message required does not exist: tf2_web_republisher, msg, TFSubscriptionActionGoal"` may be related to my issue #175

Sunshengjin, 2021-07-15:
Hi, I also have a problem about tfw_web_republisher. I found that in ros3djs, all display plug-ins that require tfClient cannot be displayed normally, such as pose, urdf, laserscan, etc.  
Here is part of  my debug information：

```
  ros2-web-bridge:Bridge Send message to subscription. +234ms
  ros2-web-bridge:Bridge JSON command received: {"op":"publish","id":"publish:/tf2_web_republisher/cancel:8","topic":"/tf2_web_republisher/cancel","msg":{"id":"goal_0.7679118211490816_1626342927244"},"latch":false} +82ms
  ros2-web-bridge:Bridge Publish a topic named /tf2_web_republisher/cancel with {"id":"goal_0.7679118211490816_1626342927244"} +1ms
  ros2-web-bridge:Bridge Response: {"op":"status","level":"error","msg":"publish: TypeError: Invalid argument: sec in Time","id":"publish:/tf2_web_republisher/cancel:8"} +1ms
  ros2-web-bridge:Bridge JSON command received: {"op":"publish","id":"publish:/tf2_web_republisher/goal:9","topic":"/tf2_web_republisher/goal","msg":{"goal_id":{"stamp":{"secs":0,"nsecs":0},"id":"goal_0.15019265547285143_1626342927560"},"goal":{"source_frames":["base_footprint","map"],"target_frame":"/odom","angular_thres":0.01,"trans_thres":0.01,"rate":30}},"latch":false} +0ms
  ros2-web-bridge:Bridge Publish a topic named /tf2_web_republisher/goal with {"goal_id":{"stamp":{"secs":0,"nsecs":0},"id":"goal_0.15019265547285143_1626342927560"},"goal":{"source_frames":["base_footprint","map"],"target_frame":"/odom","angular_thres":0.01,"trans_thres":0.01,"rate":30}} +0ms
```
  **ros2-web-bridge:Bridge Response: {"op":"status","level":"error","msg":"publish: TypeError: Invalid argument: sec in Time","id":"publish:/tf2_web_republisher/cancel:8"} +1ms**
This line seems to say that there is a problem with the Time format in tf2_web_republisher.
Is there any progress on this issue?

Thanks！

minggangw, 2021-07-16:
I think the root cause is that the [actionlib](https://github.com/RobotWebTools/roslibjs/tree/develop/src/actionlib) is implemented by the roslibjs library for ROS1, while for ROS2, the actionlib is implemented by the [rcl](https://github.com/ros2/rcl/tree/master/rcl_action) and exposed through different clients, e.g. rclpy/rclnodejs.

Unfortunately, the ros2-web-bridge doesn't expose the actionlib feature because rosbridge v2 protocol doesn't support the actionlib.

SomaGallai, 2021-08-09:
Is there a work-around for this issue?